vk2him/Enphase-Envoy-mqtt-json #3: Can't get launchctl working

sddgit, 2021-10-08:
I hope it's ok to ask here for help. I can run the script very successfully from MacOS Terminal, using 

`python3 /users/serveruser/Envoy_scripts/envoy_mqtt_script.py`

but I'm having a lot of trouble getting it to run via launchctl. My plist is:

```<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Disabled</key>
	<false/>
	<key>EnvironmentVariables</key>
	<dict>
		<key>PATH</key>
		<string>/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Library/Apple/usr/bin:/usr/local/sbin</string>
	</dict>
	<key>KeepAlive</key>
	<true/>
	<key>Label</key>
	<string>envoy</string>
	<key>ProgramArguments</key>
	<array>
		<string>/usr/bin/python3</string>
		<string>/users/serveruser/Envoy_scripts/envoy_mqtt_script.py</string>
	</array>
	<key>RunAtLoad</key>
	<true/>
	<key>StandardErrorPath</key>
	<string>/Users/serveruser/Library/logs/Envoy mqtt script/stderr.log</string>
	<key>StandardOutPath</key>
	<string>/Users/serveruser/Library/logs/Envoy mqtt script/stdout.log</string>
</dict>
</plist>
```
When I load it, the stderr.log shows

```Traceback (most recent call last):
  File "/users/serveruser/Envoy_scripts/envoy_mqtt_script.py", line 12, in <module>
    import requests
ModuleNotFoundError: No module named 'requests'
```

Now when I first started playing with this, the "import" module was missing on my system, and I installed it and all was well  via Terminal (or Python3 Launcher). But the problem is back when I use launchctl. Any ideas?

vk2him, 2021-10-08:
It sounds like you've installed the requests module as a different user that the lauchctl user is using and it can't find it.

Can you confirm when you installed the requests module that you were logged in as /users/serveruser ? 

Which user are you logged in as when you can successfully run the script from terminal - is it serveruser?

What does "pip3 show requests" display when you are able to run it from terminal - this will show where requests has been installed.

Hopefully this might give you some clues. Cheers

sddgit, 2021-10-08:
Thank you for trying to help.

Everything I've done has been under serveruser. I first of all installed Python3, then pip via `curl https://bootstrap.pypa.io/get-pip.py -o get-pip.py`, then `python3 get-pip.py` (got those instructions from a web site somewhere!). Then I followed your instructions `pip install paho-mqtt`, and running the script gave me the error about missing the "requests" module. So I installed that using `pip install requests`, and the script ran fine in Terminal, but doesn't using launchctl.

`pip3 show requests` (or `pip show requests`) gives:

```Name: requests
Version: 2.26.0
Summary: Python HTTP for Humans.
Home-page: https://requests.readthedocs.io
Author: Kenneth Reitz
Author-email: me@kennethreitz.org
License: Apache 2.0
Location: /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages
Requires: urllib3, charset-normalizer, certifi, idna
Required-by: 
```
Thanks again.

sddgit, 2021-10-09:
Fixed it. Instead of loading Python3 from /usr/bin, I now load it from /usr/local/bin, and everything's fine. I have absolutely no idea why it exists in both places!

vk2him, 2021-10-09:
Great to see that you fixed it. Macs really are a royal pain in the rear lol, I’ve had so many issues over the years cause it’s not a real Unix system and Apple do things differently . I’m seriously considering dumping mine and going for a Linux system. I might enable bootcamp on the Mac and see if I can just run Mint or another similar OS.

sddgit, 2021-10-09:
That gets `/Library/Frameworks/Python.framework/Versions/3.9/bin/python3`. Completely different again, but kind of matches where the “requests” module is installed, as above. I got the `usr/local/bin/python3` by starting the Python Launcher that was installed when I installed Python3, and that defaulted to that path. I did end up installing Xcode command line tools at some stage too to fix something, but can’t for the life of me remember why! No idea what’s going on…

vk2him, 2021-10-09:
Interesting - which python3 reports `/usr/bin/python3` for me - which python reports `/System/Library/Frameworks/Python.framework/Versions/2.7/bin/python2.7`
I hate MACOS lol

sddgit, 2021-10-09:
But what’s in your `/usr/bin/` folder? Is Python3 an alias to Python3 in the Frameworks path or an executable? In my `/usr/bin/` Python is an alias to Python (2) in the frameworks path, but Python3 is an executable (but doesn’t work properly). Do you see Python3 (alias) in `/usr/local/bin`? All very confusing. To be fair, the placements would be down to the installer wouldn’t they? I used 3.9 from [https://www.python.org/downloads/](url). I presume the installer sets up the aliases to point to the latest release installed.

sddgit, 2021-10-10:
Well mine’s not that pretty either, and I’ve only just installed Python!

sddgit, 2021-10-13:
Nice. Horses for courses!