vk2him/Enphase-Envoy-mqtt-json #4: Floods of connection errors on occasion

sddgit, 2021-10-10:
I have the script running on a Mac via launchctl. I have added 

```	
	<key>StandardErrorPath</key>
	<string>/Users/serveruser/Library/logs/Envoy mqtt script/stderr.log</string>
	<key>StandardOutPath</key>
	<string>/Users/serveruser/Library/logs/Envoy mqtt script/stdout.log</string>
```

to the plist. I’ve noticed that stdout.log will sometimes have hundreds of connection error messages like this:

```
Exception fetching stream data: HTTPConnectionPool(host='192.168.0.251', port=80): Read timed out.
Exception fetching stream data: HTTPConnectionPool(host='192.168.0.251', port=80): Max retries exceeded with url: /stream/meter (Caused by ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0x7fb33a96eb50>, 'Connection to 192.168.0.251 timed out. (connect timeout=5)'))
Exception fetching stream data: HTTPConnectionPool(host='192.168.0.251', port=80): Max retries exceeded with url: /stream/meter (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fb33a96e370>: Failed to establish a new connection: [Errno 64] Host is down'))
Exception fetching stream data: HTTPConnectionPool(host='192.168.0.251', port=80): Max retries exceeded with url: /stream/meter (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fb33a964b50>: Failed to establish a new connection: [Errno 64] Host is down'))
Exception fetching stream data: HTTPConnectionPool(host='192.168.0.251', port=80): Max retries exceeded with url: /stream/meter (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fb33a9640a0>: Failed to establish a new connection: [Errno 64] Host is down'))
```

I'm not sure what the timespan between the messages is, because there are no timestamps. I'm wondering whether anyone sees these errors on their system. My Envoy is currently connected wirelessly, with what is says is a couple of bars of reception. For clarity, the errors also happen running the script directly in a Terminal window. Thank you.

vk2him, 2021-10-10:
Wireless connection isn't a good idea for error free operation - when I ran the script from terminal for weeks during testing, I'd get an occasional error like that but not hundreds at a time. My Envoy is hardwire connected. I'd guess the number of errors you get are related to wifi signal strength - you might consider testing with a repeater to see if the errors reduce/go away?

sddgit, 2021-10-10:
Thanks.  Yes, I’ll rig something up to try to improve signal strength, or get an Ethernet cable run. Since this script adds so much to the visibility of solar systems, it’ll be worth the effort.

sddgit, 2021-10-18:
Thanks very much for doing that. I'll monitor to see if I can pick any patterns.

I do notice that even going to the Envoy's web page from a browser is sometimes very slow (even though the script is running fine at the time). How does yours respond?

Thanks for the help.

closed 2021-11-01 by vk2him

sddgit, 2021-11-01:
For completeness, my issue seems to have settled down now. No errors for a few days, mainly since I had a new Ethernet cable run. Still get slow response accessing the Envoy's web page sometimes though. Not sure if that's normal.