vk2him/Enphase-Envoy-mqtt-json #5: Struggling to get this to work

immesys, 2021-11-07:
I am getting 401 Authorization Required when the requests gets made to the stream endpoint.

I can reproduce with curl as well (installer password is generated using envoy serial number):

Doing this

```
curl --insecure --digest --user installer:123456 https://192.168.1.115/stream/meter
```

Gives me back this:

```

<html>
<head><title>401 Authorization Required</title></head>
<body>
<center><h1>401 Authorization Required</h1></center>
<hr><center>Redirecting to <a id='link' href=''></a></center>
<script>
const redirect_url = window.location.protocol + '//' + window.location.host + '/home';
function redirect() { window.location.href = redirect_url; }
window.onload = function() {
    const link = document.getElementById('link');
    link.href = redirect_url;
    link.text = redirect_url;
    setTimeout(redirect, 500);
}
</script>
</body>
</html>
```

My envoy is running firmware D7.0.66 (ddd0d7).

Is your envoy running the same firmware, but it works? Do you not have issues with getting redirected to HTTPS and having a self-signed certificate cause issues? I noticed you have modified your code recently, so I assumed you are still using this code successfully.

vk2him, 2021-11-07:
My envoy gives that error when trying to connect to https  - I connect using http only. My Envoy details:
Envoy-S-Metered-EU
Software Version 5.0.55

I stumbled across this thread recently while programming this code - is this what you're seeing?

Mine is working fine - I modded the code recently for error logging as a user had connectivity issues via wifi.

[https://community.enphase.com/s/question/0D53m00006ySLuR/unimpressed-with-loss-of-local-api-connectivity-to-envoys](url)

vk2him, 2021-11-07:
From the above thread - a few comments are:

"The internal API that everyone has lost access to is - http://x.x.x.x/api/v1/production

As mentioned it now expects a token or Enphase credentials, and forces to https.

7.x.x uses new authentication method and is installed on installation with Ensemble Devices"

So it sounds like you are affected by this - 7.x.x can't use this code unfortunately.

immesys, 2021-11-07:
Ah..

Well. That's unfortunate. That is what I am seeing.

I'm doing my own digging around how to get things to work with the new envoy firmware, I'll let you know if I find out a way to get this working again.

Feel free to close this issue, but you may want to leave it open in case others have the same problem

vk2him, 2021-11-07:
Ok thanks, yes please advise if you find anything. I’ve renamed this issue to better describe the issue

immesys, 2021-11-19:
@vk2him here is a writeup of my findings, that I wrote for a different scraper project: https://github.com/dlmcpaul/EnphaseCollector/issues/22#issuecomment-973656321

vk2him, 2021-11-19:
@immesys - Thanks for that Michael - that's really helpful. The other day my Envoy was upgraded without my approval by Enphase - I'm now on 7.x.x and the integration broke. I was able to view the steam in a browser by creating a token, however the stream only outputs every 30-40 seconds - it seems to send a page, then wait 30-40 seconds before dumping another page. 
Maybe you could assist please - can you get your token then visit https://envoy.local/stream/meter and advise if you see the same 30-40 second delay?

immesys, 2021-11-19:
I did actually see the same behavior when first trying out the stream/meter endpoint. I thought maybe it was a bug on my side with line buffering or something like that. Guess that's new then? I haven't found a way around that, I just focused on the production.json endpoint for now. Let me know if you find out how to fix that, it'd be neat to have more real-time data

vk2him, 2021-11-19:
Yeah, I've tried a few browsers on a few machines - with the same page at a time output. It might be something "they" have to fix ....

vk2him, 2022-02-16:
Update - I am still waiting for Enphase to fix the 30+second delay with the  https://envoy.local/stream/meter json stream. They have released a token system that negates the need for the installer password, however that method doesn't currently work for the  https://envoy.local/stream/meter stream.

vk2him, 2022-03-28:
Update - 7.x.x firmware is still a hot mess - most API URLs fail with 504 errors. In desperation I contacted Enphase Support via telephone and requested they downgrade my Envoy to version 5.x.x ... after placing me on hold for a few minutes the operator confirmed she could do this and went ahead and did so. I also requested that i remain on 5.x.x until I specifically requested an upgrade - again she confirmed my Envoy will remain at 5.x.x. My script is now running again perfectly.

parallelsys, 2022-10-28:
@vk2him Do you need somebody willing to test and debug?

vk2him, 2022-10-28:
@parallelsys - I'm not sure how to get 7.x.x working although there's now a few examples available of how to pass the required credentials, but I haven't investigated further. You're welcome to fork and test - the main program shouldn't need much modification, it's just the initial and ongoing authentication to get connected. Good luck! Cheers

parallelsys, 2022-10-28:
I'll give it a go.

del13r, 2022-10-30:
Step 1
I did some experimenting with postman and then cURL and found how to generate the session_id.

Login and generate session_id via cURL

```
curl https://enlighten.enphaseenergy.com/login/login.json \
-d 'user[email]=my@emailaddress.com&user[password]=MyPassword' \
| jq -r '.session_id'
```

Results

`32alphanumericcharacterSession_ID`

Step 2

cURL command that generates the JWT using the session ID from Step 1.

```
curl 'https://enlighten.enphaseenergy.com/entrez-auth-token?serial_num=myserialnumber' \
  -H 'cookie: _enlighten_4_session=32alphanumericcharacterSession_IDFromStep1'
```

Results

`{"generation_time":1660805107,"token":"Stringof408Characters","expires_at":1692341107}`

The token successfully generated the above json formatted output that contained a 408 character token.

parallelsys, 2022-10-30:
Well done @del13r

eleventhaus, 2022-11-02:
> Update - 7.x.x firmware is still a hot mess - most API URLs fail with 504 errors. In desperation I contacted Enphase Support via telephone and requested they downgrade my Envoy to version 5.x.x ... [...]

What firmware did she downgrade you to and do you recommend downgrading? Have there been any issues since? I'm debating whether I should also downgrade.

vk2him, 2022-11-02:
I was put back to D5.0.62 (5e57a9) and I have had zero issues since. My understanding is that V7 is ONLY needed if you have batteries ...

eleventhaus, 2022-11-03:
> I was put back to D5.0.62 (5e57a9) and I have had zero issues since. My understanding is that V7 is ONLY needed if you have batteries ...

I have batteries :(

parallelsys, 2022-11-12:
It seems that there is a working version

vk2him, 2023-03-24:
I'm not planning to modify for v7.x for a few reasons:

1. I don't have v7.x - they "upgraded" my envoy to v7.x a while ago without my permission and I had to ask them to revert me permanently back to v5.x - see next point
2. For the week or so I had v7.x, I used the method described in your link and found that most v5 data endpoints don't work, or they work with issues. I was unable to get a 1 second stream from v7.x and it seems the person in the link you gave is only able to provide 30 second updates. There's a long thread on the Enphase forum group about this - they promised over 18 month ago to fix those issues but never did and haven't provided any timeline of if/when it will ever work
3. That same thread discusses the fact that Enphase are effectively locking us out of our systems by using the Auth tokens - the tokens need to be renewed and if Enphase go out of business then we will never have access. Enphase said  18 months ago they will issue lifetime tokens, but they haven't said any more since then despite hundreds of people asking for updates
4. I'm not planning to ever update to v7, even if they fix it and issue lifetime tokens. V5 does all I need, I don't have batteries
5. People are welcome to fork this repo and get it working on v7 if they want - but they won't be able to get 1 second updates until Enphase fix the firmare.
Cheers

Matthew1471, 2023-06-05:
>     5. People are welcome to fork this repo and get it working on v7 if they want - but they won't be able to get 1 second updates until Enphase fix the firmare.

If someone else wants to fork but needs either a library to do the auth or want documentation these links may help:

Example: https://github.com/Matthew1471/Enphase-API/blob/main/Python/examples/gateway_database_meters.py

API Documentation: https://github.com/Matthew1471/Enphase-API/blob/main/Documentation/IQ%20Gateway%20API/Stream/Meter.adoc

niggles1973, 2023-07-06:
had my envoy updated to V7.6.175 not happy ...
anyway; I generated a token and have this 
curl -f -k -H "Accept: application/json" -H "Authorization: Bearer ALongToken -X GET https://192.168.1.x/ivp/meters/readings 
returning 1 second readings so I think @vk2him concerns about update intervals has been addressed (I agree with everything else though ..)
I have no idea how to fork in GitHub and have never used python so say a little prayer for me as I dabble in the code

networkerict, 2023-07-06:
Mine was updated to the same D7 release last night as well.... sigh.... 
If anyone with python knowledge is able to fork this release and make it work for D7 firmware.... PLEASE DO SO!

niggles1973, 2023-07-06:
OK, I'm a little bit further but two steps backwards

while 
curl -f -k -H "Accept: application/json" -H "Authorization: Bearer ALongToken -X GET https://envoy.local/ivp/meters/readings
does return 1 second updates on each seperate GET request; its not a stream

nigelbarling@Nigels-Mac-mini-5 ~ % curl -f -k -H "Accept: application/json" -H "Authorization: Bearer ALongToken" -X GET https://192.168.1.8/stream/meter  
returns 401/Forbidden

forums are saying that V7.6.175 has a "bug" and that /steam/meter is only available to an installer token with "oh we will fix that in the next release"

del13r, 2023-07-06:
Yep, on v5, you still need installer:password to access http://envoy.local/stream/meter

here is 2 seconds worth of data from v5
[image]

vk2him, 2023-07-06:
> OK, I'm a little bit further but two steps backwards
> 
> while curl -f -k -H "Accept: application/json" -H "Authorization: Bearer ALongToken -X GET https://envoy.local/ivp/meters/readings does return 1 second updates on each seperate GET request; its not a stream
> 
> curl -f -k -H "Accept: application/json" -H "Authorization: Bearer ALongToken" -X GET https://192.168.1.8/stream/meter returns 401/Forbidden
> 
> forums are saying that V7.6.175 has a "bug" and that /steam/meter is only available to an installer token with "oh we will fix that in the next release" - the code as it is needs the the stream version (unless its rewritten to query repeatedly)

Your methodology is sound - before you can write the code you need to verify the commands will work from terminal. You're seeing a variation on the "hot mess' post I made above 15 months ago - one step forward, twosteps back indeed. 

Wishing you luck - unless you need 7.0 for batteries, you can always request Enphase permanently downgraded your firmware to 5.xx as I did, then this issue goes away.
Cheers

niggles1973, 2023-07-06:
I’m getting a 5P/system controller 3 in a couple of
Months so rolling back is just going to postpone 7.x - May as well attack it now

niggles1973, 2023-07-07:
OK, I've got a good enough workaround (but not pretty when compared to the stream solution) - uses the equivalent of 
 curl -f -k -H "Accept: application/json" -H "Authorization: Bearer ALongToken -X GET https://envoy.local/ivp/meters/readings  
in  configuration.yaml with a rest template 
 
rest:
  - headers:
      Authorization: Bearer ***TheToken****
    verify_ssl: False
    scan_interval: 1
    resource: https://envoy.local/ivp/meters/readings    
    sensor:
      - name: "the_total_production"
        unique_id: the_total_production
        value_template: '{% if is_state("sun.sun", "below_horizon")%}0{%else%} {{ value_json[0].activePower | float }}{%endif%}'
        device_class: power
        unit_of_measurement: W
        state_class: measurement
      - name: "the_total_consumption"
        unique_id: the_total_consumption
        value_template: '{% if is_state("sun.sun", "below_horizon")%}value_json[1].activePower{%else%}{{ value_json[1].activePower - value_json[0].activePower  | float }}{%endif%}'
        device_class: power
        unit_of_measurement: W

vk2him, 2023-07-07:
Interesting- what data does https://envoy.local/ivp/meters/readings provide, can you dump the output here?

niggles1973, 2023-07-07:
readings gives the most relevant metrics via a single api call that be used to duplicate the enphase integration - but you have to hit the envoy.local every second; I occassionally get to many sessions - I'm trying to work out what the rate limit is (or I'm just hitting it too hard while debugging from multiple IP addresses)
```
[
    {
        "eid": 704643328,
        "timestamp": 1688747134,
        "actEnergyDlvd": 53457940.250,
        "actEnergyRcvd": 102882.138,
        "apparentEnergy": 64452408.104,
        "reactEnergyLagg": 11447952.939,
        "reactEnergyLead": 1570264.360,
        "instantaneousDemand": -15.184,
        "activePower": -15.184,
        "apparentPower": 1433.530,
        "reactivePower": 1419.968,
        "pwrFactor": -0.010,
        "voltage": 716.121,
        "current": 6.006,
        "freq": 50.000,
        "channels": [
            {
                "eid": 1778385169,
                "timestamp": 1688747134,
                "actEnergyDlvd": 18007780.121,
                "actEnergyRcvd": 32493.281,
                "apparentEnergy": 21780901.616,
                "reactEnergyLagg": 3956061.302,
                "reactEnergyLead": 330755.947,
                "instantaneousDemand": -4.958,
                "activePower": -4.958,
                "apparentPower": 496.173,
                "reactivePower": 490.175,
                "pwrFactor": -0.015,
                "voltage": 237.923,
                "current": 2.086,
                "freq": 50.000
            },
            {
                "eid": 1778385170,
                "timestamp": 1688747134,
                "actEnergyDlvd": 17875774.043,
                "actEnergyRcvd": 12976.762,
                "apparentEnergy": 21539705.521,
                "reactEnergyLagg": 3804013.582,
                "reactEnergyLead": 599499.184,
                "instantaneousDemand": -2.110,
                "activePower": -2.110,
                "apparentPower": 476.495,
                "reactivePower": 472.980,
                "pwrFactor": 0.000,
                "voltage": 239.413,
                "current": 1.990,
                "freq": 50.000
            },
            {
                "eid": 1778385171,
                "timestamp": 1688747134,
                "actEnergyDlvd": 17574386.086,
                "actEnergyRcvd": 57412.095,
                "apparentEnergy": 21131800.966,
                "reactEnergyLagg": 3687878.056,
                "reactEnergyLead": 640009.230,
                "instantaneousDemand": -8.115,
                "activePower": -8.115,
                "apparentPower": 460.863,
                "reactivePower": 456.814,
                "pwrFactor": -0.016,
                "voltage": 238.785,
                "current": 1.929,
                "freq": 50.000
            }
        ]
    },
    {
        "eid": 704643584,
        "timestamp": 1688747134,
        "actEnergyDlvd": 22984313.125,
        "actEnergyRcvd": 36889465.180,
        "apparentEnergy": 75005727.629,
        "reactEnergyLagg": 690026.580,
        "reactEnergyLead": 25946784.470,
        "instantaneousDemand": 3958.407,
        "activePower": 3958.407,
        "apparentPower": 5209.561,
        "reactivePower": -2945.972,
        "pwrFactor": 0.760,
        "voltage": 716.550,
        "current": 21.809,
        "freq": 50.125,
        "channels": [
            {
                "eid": 1778385425,
                "timestamp": 1688747134,
                "actEnergyDlvd": 8775639.959,
                "actEnergyRcvd": 12568251.034,
                "apparentEnergy": 25610630.935,
                "reactEnergyLagg": 199715.930,
                "reactEnergyLead": 7406932.417,
                "instantaneousDemand": 1361.722,
                "activePower": 1361.722,
                "apparentPower": 1694.904,
                "reactivePower": -832.508,
                "pwrFactor": 0.806,
                "voltage": 238.052,
                "current": 7.118,
                "freq": 50.125
            },
            {
                "eid": 1778385426,
                "timestamp": 1688747134,
                "actEnergyDlvd": 7471792.993,
                "actEnergyRcvd": 11470877.359,
                "apparentEnergy": 24964774.206,
                "reactEnergyLagg": 260425.150,
                "reactEnergyLead": 9708022.585,
                "instantaneousDemand": 1205.898,
                "activePower": 1205.898,
                "apparentPower": 1763.992,
                "reactivePower": -1181.869,
                "pwrFactor": 0.681,
                "voltage": 239.399,
                "current": 7.370,
                "freq": 50.125
            },
            {
                "eid": 1778385427,
                "timestamp": 1688747134,
                "actEnergyDlvd": 6736880.174,
                "actEnergyRcvd": 12850336.786,
                "apparentEnergy": 24430322.489,
                "reactEnergyLagg": 229885.501,
                "reactEnergyLead": 8831829.467,
                "instantaneousDemand": 1390.786,
                "activePower": 1390.786,
                "apparentPower": 1750.664,
                "reactivePower": -931.595,
                "pwrFactor": 0.794,
                "voltage": 239.099,
                "current": 7.321,
                "freq": 50.125
            }
        ]
    }
```

vk2him, 2023-07-07:
Thanks for that - looks like the similar data from /stream but in a different format. The downside of course is that it's not a stream so you need to send a request and hammering the envoy every second may cause an overflow. Did you just post a question on the Enphase Community as Barling_114 asked a similar question? It's a shame the /stream feed is still broken on V7 - it didn't work nearly two years ago when I had V7 temporarily ...

niggles1973, 2023-07-07:
yeah I'm working both angles ... (this and trying to get /stream back accessible by the owner token)

helderfmf, 2023-07-10:
I have done it, see here https://github.com/helderfmf/Enphase-Envoy-mqtt-json. 
Ignore the Freeds stuff, I'm publishing to 2 different topics over mqtt.
Need to implement the automatic generation of the token on startup, and error handling, as there is none at the moment. But it works. 
Also the sensors on the HA changed:
```dockerfile
mqtt:
  ###############
  #   SENSOR    #
  ###############
  sensor:
    ###############
    #    Envoy    #
    ###############
    - name: envoy mqtt consumption
      state_topic: "envoy/json"
      value_template: '{{ value_json["consumption"][0]["wNow"] | round(0) | int(0)}}'
      unique_id: envoy_mqtt_consumption
      qos: 0
      unit_of_measurement: "W"
      state_class: measurement
      device_class: power
    - name: envoy mqtt voltage
      state_topic: "envoy/json"
      value_template: '{{ value_json["consumption"][0]["rmsVoltage"] | round(0) | int(0)}}'
      unique_id: envoy_mqtt_voltage
      qos: 0
      unit_of_measurement: "V"
      state_class: measurement
      device_class: voltage
    - name: envoy mqtt current
      state_topic: "envoy/json"
      value_template: '{{ value_json["consumption"][0]["rmsCurrent"] | round(2)}}'
      unique_id: envoy_mqtt_current
      qos: 0
      unit_of_measurement: "A"
      state_class: measurement
      device_class: current
    - name: envoy mqtt power factor
      state_topic: "envoy/json"
      value_template: '{{ value_json["consumption"][0]["pwrFactor"] | round(2)}}'
      unique_id: envoy_mqtt_power_factor
      qos: 0
      unit_of_measurement: "%"
      state_class: measurement
      device_class: power_factor
```

vk2him, 2023-07-10:
> I have done it, see here https://github.com/helderfmf/Enphase-Envoy-mqtt-json. Ignore the Freeds stuff, I'm publishing to 2 different topics over mqtt. Need to implement the automatic generation of the token on startup, and error handling, as there is none at the moment. But it works.  ```

Hey @helderfmf , well done!  I took a look at your code yesterday and was wondering how you were going with it.

I see you're using production.json  - what data does that show on V7 and is it a stream with one second resolution similar to /stream/meter?

helderfmf, 2023-07-10:
This is a reading I got just now:
```
{
    "consumption": [
        {
            "activeCount": 1,
            "apprntPwr": 1232.878,
            "measurementType": "total-consumption",
            "pwrFactor": -0.01,
            "reactPwr": 10.129,
            "readingTime": 1688988057,
            "rmsCurrent": -5.199,
            "rmsVoltage": 237.148,
            "type": "eim",
            "vahLifetime": 8290045.344,
            "vahToday": 6501.344,
            "varhLagLifetime": 112923.965,
            "varhLagToday": 9.965,
            "varhLeadLifetime": 2276016.539,
            "varhLeadToday": 2979.539,
            "whLastSevenDays": 20581.0,
            "whLifetime": 3882261.863,
            "whToday": 1242.0,
            "wNow": -18.304
        },
        {
            "activeCount": 1,
            "apprntPwr": 1232.878,
            "measurementType": "net-consumption",
            "pwrFactor": -0.01,
            "reactPwr": 0.0,
            "readingTime": 1688988057,
            "rmsCurrent": -5.199,
            "rmsVoltage": 237.148,
            "type": "eim",
            "vahLifetime": 8227015.712,
            "vahToday": 0,
            "varhLagLifetime": 112142.171,
            "varhLagToday": 0,
            "varhLeadLifetime": 2276012.831,
            "varhLeadToday": 0,
            "whLastSevenDays": 0,
            "whLifetime": 3882261.8,
            "whToday": 0,
            "wNow": -18.304
        }
    ],
    "production": [
        {
            "activeCount": 0,
            "readingTime": 0,
            "type": "inverters",
            "whLifetime": 0,
            "wNow": 0
        },
        {
            "activeCount": 1,
            "apprntPwr": 0.0,
            "measurementType": "production",
            "pwrFactor": 0.0,
            "reactPwr": -0.0,
            "readingTime": 1688988057,
            "rmsCurrent": -0.0,
            "rmsVoltage": 237.23,
            "type": "eim",
            "vahLifetime": 63029.632,
            "vahToday": 69.632,
            "varhLagLifetime": 781.794,
            "varhLagToday": 0.794,
            "varhLeadLifetime": 3.709,
            "varhLeadToday": 0.709,
            "whLastSevenDays": 0.063,
            "whLifetime": 0.063,
            "whToday": 0.063,
            "wNow": 0.0
        }
    ],
    "storage": [
        {
            "activeCount": 0,
            "readingTime": 0,
            "state": "idle",
            "type": "acb",
            "whNow": 0,
            "wNow": 0
        }
    ]
}
``` 
In HA i got a latency of 2-4s between readings.

del13r, 2023-07-10:
Give it a try for 24 hours and see how the envoy likes being queried that often. I tried every 5 seconds and saw lengthy times where envoy would not respond at all

niggles1973, 2023-07-10:
Production.Json is too slow to respond and sometimes takes up to 60 seconds
or more (I’ve got 84 microinverters so might be my system only) -
https://envoy.local/ivp/meters/readings
Is consistently 1 second

niggles1973, 2023-07-10:
I’ve been hitting https://envoy.local/ivp/meters/readings
Every second for 48 hours now;  production.Json was timing out repeatedly in only 12 hours of Testing and was averaging 3 seconds or so.  Testing is with the home assistant rest template.

del13r, 2023-07-10:
unfortunately i didnt have as good results as you with 5 second polling.
I have found 10 seconds to be much less problematic.
graph of dropouts when using 5 second polling:
[image]

helderfmf, 2023-07-10:
I'm using only as a meter from the grid, my inverter is from Huawei, and the solar installer instead of using some cheaper meter, installed this one that is a overkill for the function it does.

So way less information to compute on each call.

Hélder Ferreira

del13r, 2023-07-10:
> I'm using only as a meter from the grid
> So way less information to compute on each call.

I'm glad you could fix your problem, but we need to be considerate of the wider community expectations that may read your posts and expect the same results.
This definitely does not fit the typical use case and your 1 poll per second solution could potentially be misleading for your average enphase user. 
You might get a lot of people asking you why it works for you but not for them.

helderfmf, 2023-07-11:
Your are using REST, I'm using the python with stream, don't know if it makes a diference.
Any way, try this:
`curl -k -H "Authorization: Bearer YourVeryLongToken" https://envoy.local/ipv/livedata/stream/ -X POST '{"enable": 1}'`
You should get a response:
```
{
"sc_stream": "enabled"
}
```
After that this counters should update, and for me they are quick:
`curl -k -H "Authorization: Bearer YourVeryLongToken" https://envoy.local/ipv/livedata/status`

helderfmf, 2023-07-11:
When I say its fast, it is really fast, @del13r can you try it?
here is the code:
```
def scrape_stream():
    n=0
    while True:
        try:
            url = 'https://%s/ivp/livedata/status' % ENVOY_HOST 
            stream = requests.get(url, auth=auth, stream=True, timeout=5, verify=False, headers=headers)
            json_string = stream.json()
            pp.pprint(json_string)
            pp.pprint(time.ctime())
            n=n+1
            pp.pprint('Run %s times' %n)
        except requests.exceptions.RequestException as e:
            print(dt_string, ' Exception fetching stream data: %s' % e)

``` 
The output, in 2 seconds 7 replies: 

```
>>> scrape_stream()
{'connection': {'auth_state': 'ok',
                'mqtt_state': 'disconnected',
                'prov_state': 'configured',
                'sc_debug': 'disabled',
                'sc_stream': 'enabled'},
 'counters': {'MqttClient_connect': 14809,
              'MqttClient_connect_Err': 14809,
              'MqttClient_connect_err': 14809,
              'MqttClient_create': 14809,
              'MqttClient_setCallbacks': 14809,
              'SSL_Keys_Create': 14809,
              'main_CfgChanged': 1,
              'main_CfgLoad': 1,
              'rest_Counters': 20,
              'rest_Meters': 23,
              'rest_Status': 3982,
              'rest_Tasks': 1,
              'rest_Test': 2,
              'sc_SendDemandRspCtrl': 1,
              'sc_SendStreamCtrl': 10,
              'sc_hdlDataPub': 11668},
 'dry_contacts': {'': {'dry_contact_id': '',
                       'dry_contact_load_name': '',
                       'dry_contact_status': 3050996,
                       'dry_contact_type': ''}},
 'meters': {'acb_agg_energy': 0,
            'acb_agg_soc': 0,
            'backup_bat_mode': 1,
            'backup_soc': 30,
            'enc_agg_energy': 0,
            'enc_agg_soc': 0,
            'gen_relay_state': 5,
            'generator': {'agg_p_mw': 0,
                          'agg_p_ph_a_mw': 0,
                          'agg_p_ph_b_mw': 0,
                          'agg_p_ph_c_mw': 0,
                          'agg_s_mva': 0,
                          'agg_s_ph_a_mva': 0,
                          'agg_s_ph_b_mva': 0,
                          'agg_s_ph_c_mva': 0},
            'grid': {'agg_p_mw': 98293,
                     'agg_p_ph_a_mw': 98293,
                     'agg_p_ph_b_mw': 0,
                     'agg_p_ph_c_mw': 0,
                     'agg_s_mva': 1217741,
                     'agg_s_ph_a_mva': 1217741,
                     'agg_s_ph_b_mva': 0,
                     'agg_s_ph_c_mva': 0},
[... 488 lines cut ...]
                     'agg_s_ph_b_mva': 0,
                     'agg_s_ph_c_mva': 0},
            'main_relay_state': 1,
            'phase_count': 1,
            'pv': {'agg_p_mw': 0,
                   'agg_p_ph_a_mw': 0,
                   'agg_p_ph_b_mw': 0,
                   'agg_p_ph_c_mw': 0,
                   'agg_s_mva': 0,
                   'agg_s_ph_a_mva': 0,
                   'agg_s_ph_b_mva': 0,
                   'agg_s_ph_c_mva': 0},
            'soc': 0,
            'storage': {'agg_p_mw': 0,
                        'agg_p_ph_a_mw': 0,
                        'agg_p_ph_b_mw': 0,
                        'agg_p_ph_c_mw': 0,
                        'agg_s_mva': 0,
                        'agg_s_ph_a_mva': 0,
                        'agg_s_ph_b_mva': 0,
                        'agg_s_ph_c_mva': 0}},
 'tasks': {'task_id': 0, 'timestamp': 0}}
'Tue Jul 11 11:25:07 2023'
'Run 7 times'
```

niggles1973, 2023-07-11:
Hi @helderfmf ,
what scope is your token? installer or owner - from V7.6.175
prompt@Nigels-Mac-mini-5 ~ % curl -k -H #VeryLongToken#  https://envoy.local/ipv/livedata/stream/ -X POST '{"enable": 1}'               

<html>
<head><title>404 Not Found</title></head>
<body>
<center><h1>404 Not Found</h1></center>
<hr><center>openresty/1.17.8.1</center>
</body>
</html>
curl: (3) URL using bad/illegal format or missing URL

helderfmf, 2023-07-11:
sorry, made a typo, NOT https://envoy.local/ipv/livedata/stream/ but https://envoy.local/ipv/livedata/stream without the ending /

del13r, 2023-07-11:
command:
curl -f -k -H "Authorization:bearer tokengoeshere" -H "Content-Type:application/json" -d "{\"enable\":1}" https://envoy.local/ivp/livedata/stream

result:
{
    "sc_stream": "enabled"
}

command:
curl -f -k -H "Authorization:bearer tokengoeshere" -H "Content-Type:application/json" https://envoy.local/ivp/livedata/status -v

Result:
*   Trying my.local.ip.v4:443...
* Connected to envoy.local (my.local.ip.v4) port 443 (#0)
* ALPN: offers h2,http/1.1
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
* TLSv1.3 (IN), TLS handshake, Server hello (2):
* TLSv1.2 (IN), TLS handshake, Certificate (11):
* TLSv1.2 (IN), TLS handshake, Server key exchange (12):
* TLSv1.2 (IN), TLS handshake, Server finished (14):
* TLSv1.2 (OUT), TLS handshake, Client key exchange (16):
* TLSv1.2 (OUT), TLS change cipher, Change cipher spec (1):
* TLSv1.2 (OUT), TLS handshake, Finished (20):
* TLSv1.2 (IN), TLS handshake, Finished (20):
* SSL connection using TLSv1.2 / ECDHE-RSA-AES256-GCM-SHA384
* ALPN: server did not agree on a protocol. Uses default.
* Server certificate:
*  subject: CN=envoy-MYSERIAL
*  start date: Jul  7 23:36:19 2023 GMT
*  expire date: Jul  6 23:36:19 2024 GMT
*  issuer: CN=envoy-MYSERIAL
*  SSL certificate verify result: self-signed certificate (18), continuing anyway.
* using HTTP/1.x
> GET /ivp/livedata/status HTTP/1.1
> Host: envoy.local
> User-Agent: curl/8.1.2
> Accept: */*
> Authorization:bearer tokengoesherre
> Content-Type:application/json
>
< HTTP/1.1 200 OK
< Server: openresty/1.17.8.1
< Date: Tue, 11 Jul 2023 10:59:21 GMT
< Content-Length: 3387
< Connection: keep-alive
< Set-Cookie: sessionId=3rSzxacePCcq2QoJx32HagCZ8nB96LkV; Secure; HttpOnly; path=/
< Strict-Transport-Security: max-age=63072000; includeSubdomains
< X-Frame-Options: DENY
< X-Content-Type-Options: nosniff
<

Thanks for the pointers, but it's not a traditional stream if you have to keep querying it.

niggles1973, 2023-07-11:
prompt@Nigels-Mac-mini-5 ~ % curl -v -k -H "Authorization: Bearer thetoken"  https://envoy.local/ipv/livedata/stream -X POST '{"enable": 1}'

*   Trying 192.168.1.8:443...
* Connected to envoy.local (192.168.1.8) port 443 (#0)
* ALPN: offers h2,http/1.1
* (304) (OUT), TLS handshake, Client hello (1):
* (304) (IN), TLS handshake, Server hello (2):
* TLSv1.2 (IN), TLS handshake, Certificate (11):
* TLSv1.2 (IN), TLS handshake, Server key exchange (12):
* TLSv1.2 (IN), TLS handshake, Server finished (14):
* TLSv1.2 (OUT), TLS handshake, Client key exchange (16):
* TLSv1.2 (OUT), TLS change cipher, Change cipher spec (1):
* TLSv1.2 (OUT), TLS handshake, Finished (20):
* TLSv1.2 (IN), TLS change cipher, Change cipher spec (1):
* TLSv1.2 (IN), TLS handshake, Finished (20):
* SSL connection using TLSv1.2 / ECDHE-RSA-AES256-GCM-SHA384
* ALPN: server did not agree on a protocol. Uses default.
* Server certificate:
*  subject: CN=envoy-122108032292
*  start date: Jul  6 07:23:48 2023 GMT
*  expire date: Jul  5 07:23:48 2024 GMT
*  issuer: CN=envoy-122108032292
*  SSL certificate verify result: self signed certificate (18), continuing anyway.
* using HTTP/1.x
> POST /ipv/livedata/stream HTTP/1.1
> Host: envoy.local
> User-Agent: curl/7.88.1
> Accept: */*
> Authorization: Bearer the token
> 
< HTTP/1.1 404 Not Found
< Server: openresty/1.17.8.1
< Date: Tue, 11 Jul 2023 11:03:22 GMT
< Content-Type: text/html
< Content-Length: 159
< Connection: keep-alive
< 
<html>
<head><title>404 Not Found</title></head>
<body>
<center><h1>404 Not Found</h1></center>
<hr><center>openresty/1.17.8.1</center>
</body>
</html>
* Connection #0 to host envoy.local left intact
* Closing connection -1
curl: (3) URL using bad/illegal format or missing URL

I read somewhere about needing to take the cookie from the curl -v output of /home and using --cookie? (Enphase community forums I think)

niggles1973, 2023-07-11:
hang on; saw your latest posting; will copy exactly as you have

del13r, 2023-07-11:
If it was a proper stream, it should say:

< Content-Type: text/event-stream

niggles1973, 2023-07-11:
This is on 7.6.175 without the / - tried both below and helderfmf's version of the command 
nigelbarling@Nigels-Mac-mini-5 ~ % curl -v -f -k -H "Authorization:bearer Mytoken" -H "Content-Type:application/json" -d "{"enable":1}" https://envoy.local/ivp/livedata/stream
*   Trying 192.168.1.8:443...
* Connected to envoy.local (192.168.1.8) port 443 (#0)
* ALPN: offers h2,http/1.1
* (304) (OUT), TLS handshake, Client hello (1):
* (304) (IN), TLS handshake, Server hello (2):
* TLSv1.2 (IN), TLS handshake, Certificate (11):
* TLSv1.2 (IN), TLS handshake, Server key exchange (12):
* TLSv1.2 (IN), TLS handshake, Server finished (14):
* TLSv1.2 (OUT), TLS handshake, Client key exchange (16):
* TLSv1.2 (OUT), TLS change cipher, Change cipher spec (1):
* TLSv1.2 (OUT), TLS handshake, Finished (20):
* TLSv1.2 (IN), TLS change cipher, Change cipher spec (1):
* TLSv1.2 (IN), TLS handshake, Finished (20):
* SSL connection using TLSv1.2 / ECDHE-RSA-AES256-GCM-SHA384
* ALPN: server did not agree on a protocol. Uses default.
* Server certificate:
*  subject: CN=envoy-122108032292
*  start date: Jul  6 07:23:48 2023 GMT
*  expire date: Jul  5 07:23:48 2024 GMT
*  issuer: CN=envoy-122108032292
*  SSL certificate verify result: self signed certificate (18), continuing anyway.
* using HTTP/1.x
> POST /ivp/livedata/stream HTTP/1.1
> Host: envoy.local
> User-Agent: curl/7.88.1
> Accept: */*
> Authorization:bearer <sensored>
> Content-Type:application/json
> Content-Length: 10
> 
< HTTP/1.1 400 Bad Request
< Server: openresty/1.17.8.1
< Date: Tue, 11 Jul 2023 11:11:57 GMT
< Content-Length: 72
< Connection: keep-alive
< Set-Cookie: sessionId=Ofyb3kmamwJH7jpwvs5KX7xXdEtw61h9; Secure; HttpOnly; path=/
* The requested URL returned error: 400
* Closing connection 0
* TLSv1.2 (OUT), TLS alert, close notify (256):
curl: (22) The requested URL returned error: 400

helderfmf, 2023-07-11:
Something is off, my version is the same as yours, https://envoy.local/info:
`<software>D7.6.175</software>`
My token was obtained at https://entrez.enphaseenergy.com/entrez_tokens:
[image]
If you start typing 10000 it should apear, if not go to https://enlighten.enphaseenergy.com/ and obtain this information of the system and gateway.

My curl commands work as I typed above:
[image]

[image]

I honestly don't know why yours are giving 404, could be 401 ou 405, but not 404.

niggles1973, 2023-07-11:
I've got an early flight; so need to stop looking; one for the weekend when I'm back home - that process for the token generation is the one I used - I got my token this last weekend - can I confirm is your account setup as an owner and not an installer?  my account is an owner account (token has a 1 year expiry)

del13r, 2023-07-11:
I tested the difference between
https://envoy.local/ivp/livedata/status
And
https://envoy.local/ivp/meters/readings

they are both json output with equal speed and latency.

The difference being that ivp/meters/readings appears to give current power and lifetime energy stats of the 2 ct clamps. 1 for panels/production and the other is net grid (grid import/grid export). 

whereas with ivp/livedata/status, you have to enable a streaming flag first which is misleading as it doesn’t appear to be actually streaming anywhere if you have to repeatedly query it. I don’t currently have batteries so my interest in this url is lower as it doesn’t provide me any extra.

I’ve written an article here

https://community.home-assistant.io/t/enphase-envoy-with-energy-dashboard/328668/662?u=del13r

helderfmf, 2023-07-12:
Just a heads up, tested https://envoy.local/ivp/meters/readings, and its as fast as the livedata, so implemented in my fork, also the automatic token generation on startup and in the case of expired token.
Sofar its running stable, getting readings in 1s frequency.
[image]

[image]

del13r, 2023-07-12:
Great effort. Much appreciated.

Just wondering, is your envoy connected to wifi or Ethernet?
I ask this as I had an issue/bug this week where my envoy repeatedly kept trying to disassociate and then Re-associate to my router several times a minute. I restarted my envoy and that didn’t fix it the issue, so I restarted my router and that seemed to fix it. So I’m unsure as to why my router and the envoy weren’t getting along at that point in time. I assumed it had something to do with how fast I was querying the envoy. Something to keep an eye out for anyway.

helderfmf, 2023-07-12:
Had it on wifi but sometimes got packet loss, wifi signal on envoy is very bad. Next to it my phone had 3 of 4 bars and envoy was missing packets and disconnect from wifi. Got a Wifi repeater and installed 2m distance from it, but even then it would miss some reports.
I have an automation in HA that notifies me if the sensor is not updated for more than 45s.
Gave up and installed the repeater next to it and passed a Ethernet cable. Problem solved.
This was all with FW 5, don't know if FW 7 improves anything. I think it's more of a hardware problem, not a software one.

vk2him, 2023-07-12:
Excellent work on this solution - well done! I "may" modify my code to have an initial test to determine the Envoy firmware - I could then use my original code for v5 and yours for v7 .

del13r, 2023-07-13:
Yep, even at 15 seconds, my envoy and router arent getting along

`13:28:07 eth1: Deauth_ind 64:33:DB:HIDDEN, status: 0, reason: Disassociated due to inactivity (4), rssi:0
13:29:14 eth1: Auth 64:33:DB:HIDDEN, status: Successful (0), rssi:0
13:29:14 eth1: Auth 64:33:DB:HIDDEN, status: Successful (0), rssi:0
13:29:14 eth1: Assoc 64:33:DB:HIDDEN, status: Successful (0), rssi:0
13:29:14 eth1: Assoc 64:33:DB:HIDDEN, status: Successful (0), rssi:0
13:29:14 eth1: Assoc 64:33:DB:HIDDEN, status: Successful (0), rssi:0
13:29:18 eth1: Auth 64:33:DB:HIDDEN, status: Successful (0), rssi:0
13:29:18 eth1: Assoc 64:33:DB:HIDDEN, status: Successful (0), rssi:0
13:30:27 eth1: Deauth_ind 64:33:DB:HIDDEN, status: 0, reason: Disassociated due to inactivity (4), rssi:0
13:33:23 eth1: Auth 64:33:DB:HIDDEN, status: Successful (0), rssi:0
13:33:24 eth1: Assoc 64:33:DB:HIDDEN, status: Successful (0), rssi:0
`

vk2him, 2023-07-13:
What router do you have? I had an Asus that worked fine but I've replaced a while ago and now run OPNsense

del13r, 2023-07-13:
I have a 10 year old Asus Router running merlin firmware.
I think I am overloading the router CPU by running too many features at once on this old router like:

- diversion ad blocker
- connmon to healthchecks.io
- openvpn server
- DNS-over-TLS (DoT)

[70 more replies]

closed 2023-11-19 by vk2him