vk2him/Enphase-Envoy-mqtt-json #7: Values not showing up in "energy" panel

timolow, 2022-05-11:
Howdy,

I installed this extension into HA, I can see the metrics being imported into HA via the entities tab, however these entities do not show up under the energy tab in HA.

config.yaml

> sensor:

  #
  # These ones are for Envoy via mqtt
  #
  - platform: mqtt
    state_topic: "/envoy/json"
    name: "mqtt_production"
    qos: 0
    unit_of_measurement: "W"
    value_template: '{% if is_state("sun.sun", "below_horizon")%}0{%else%}{{ value_json["production"]["ph-a"]["p"]  | int(0) }}{%endif%}'
    state_class: measurement
    device_class: power

  - platform: mqtt
    state_topic: "/envoy/json"
    value_template: '{{ value_json["total-consumption"]["ph-a"]["p"] }}'
    name: "mqtt_consumption"
    qos: 0
    unit_of_measurement: "W"
    state_class: measurement
    device_class: power

  - platform: template
    sensors:
      exporting:
        friendly_name: "Current MQTT Energy Exporting"
        value_template: "{{ [0, (states('sensor.mqtt_production') | int(0) - states('sensor.mqtt_consumption') | int(0))] | max }}"
        unit_of_measurement: "W"
        icon_template: mdi:flash
      importing:
        friendly_name: "Current MQTT Energy Importing"
        value_template: "{{ [0, (states('sensor.mqtt_consumption') | int(0) - states('sensor.mqtt_production') | int(0))] | max }}"
        unit_of_measurement: "W"
        icon_template: mdi:flash
      solarpower:
        friendly_name: "Solar MQTT Power"
        value_template: "{{ states('sensor.mqtt_production')}}"
        unit_of_measurement: "W"
        icon_template: mdi:flash

vk2him, 2022-05-11:
Hey Tim, to use these for the Energy Dashboard you'll need to add this to your configuration.yaml - be sure to change the sensor numbers to not conflict with any you may already have setup. I'll update the documentation with this. let me know if this solves your issues?

```
sensor 10:
  - platform: integration
    source: sensor.solarpower
    name: energy_solar
    method: left
    #   unit: kWh
    unit_prefix: k
    unit_time: h
    round: 2
sensor 11:
  - platform: integration
    source: sensor.exporting
    name: energy_exporting
    method: left
    #    unit: kWh
    unit_prefix: k
    unit_time: h
    round: 2
sensor 12:
  - platform: integration
    source: sensor.importing
    name: energy_importing
    method: left
    #    unit: kWh
    unit_prefix: k
    unit_time: h
    round: 2
```

Now add  those new sensors to the Energy Dashboard:

[image]

timolow, 2022-05-11:
Howdy!

I went ahead and added the sensors to config.yaml, restarted, they show up under entities however they do not show up under the dropdown box for the energy dashboard.  Trying to manually insert them will not allow me to save.

timolow, 2022-05-11:
Uncommenting unit: kWh and restarting seems to have done the trick!

timolow, 2022-05-11:
Might have been short lived, I don't see any data populated nor do I see the sensors anymore in entities.  I do see this warning after adding the sensors to the energy dashboard.

```
Entity not defined
Check the integration or your configuration that provides:
sensor.energy_importing
sensor.energy_exporting
```

vk2him, 2022-05-12:
Ahh - I see the problem - you need to make a change to say that those sensors are set as  device_class: energy

Firstly, comment out the unit: kWh for those sensors as it's no longer valid.

Next, please check if you have this code in configuration.yaml - be sure that it is formatted exactly as shown including the indents. Hopefully you already have it there, if not, add it.

```
homeassistant:
  customize: !include customize.yaml
```

Now edit customize.yaml - if it doesn't exist then create it. Add the following lines:

```
sensor.energy_exporting:
  device_class: energy
sensor.energy_solar:
  device_class: energy
sensor.energy_importing:
  device_class: energy
```

After restarting Home Assistant, go back and reselect the sensors in Energy Dashboard - they should not have errors afterwards and it should match the screenshot above. Note that it will take around 2 hours before the Energy dashboard will display anything, so be patient. Please let me know if this fixes it?

timolow, 2022-05-12:
Perfect!  That did it for me.  Thank you for all the hard work you put into this plugin and helping me get this sorted!

vk2him, 2022-05-12:
Thanks @timolow, I'm glad it's working ok for you now - I'll update the Readme file to include this

vk2him, 2022-05-16:
@timolow - thanks for the donation, it wasn't expected but very much appreciated - cheers