vk2him/Enphase-Envoy-mqtt-json #8: Data from HA energy tab does not reflect enlighten data.

timolow, 2022-05-14:
Howdy,

Using the energy plugin from HA and your enphase-mqtt plugin the numbers to not align to what enphase enlighten data shows.

Example:
May 13th Production/Consumption
HA: 18.9 kWH/24.5kWH
Enlighten: 40.2kWH/91.6kWH

May 12th
HA:21.6kWh/26.4kWh
Enlighten:  44.4kWh/94.9kWh

Here is my config minus the customization.yaml file you provided me with in another issue.

```
sensor:
  - platform: mqtt
    state_topic: "/envoy/json"
    name: "mqtt_production"
    qos: 0
    unit_of_measurement: "W"
    value_template: '{% if is_state("sun.sun", "below_horizon")%}0{%else%}{{ value_json["production"]["ph-a"]["p"]  | int(0) }}{%endif%}'
    state_class: measurement
    device_class: power

  - platform: mqtt
    state_topic: "/envoy/json"
    value_template: '{{ value_json["total-consumption"]["ph-a"]["p"] }}'
    name: "mqtt_consumption"
    qos: 0
    unit_of_measurement: "W"
    state_class: measurement
    device_class: power

  - platform: mqtt
    state_topic: "/envoy/json"
    name: "mqtt_power_factor"
    qos: 0
    unit_of_measurement: "%"
    value_template: '{{ value_json["total-consumption"]["ph-a"]["pf"] }}'
    state_class: measurement
    device_class: power_factor

  - platform: mqtt
    state_topic: "/envoy/json"
    name: "mqtt_voltage"
    qos: 0
    unit_of_measurement: "V"
    value_template: '{{ value_json["total-consumption"]["ph-a"]["v"] }}'
    state_class: measurement
    device_class: voltage
sensor:

  #
  # These ones are for Envoy via mqtt
  #
  - platform: mqtt
    state_topic: "/envoy/json"
    name: "mqtt_production"
    qos: 0
    unit_of_measurement: "W"
    value_template: '{% if is_state("sun.sun", "below_horizon")%}0{%else%}{{ value_json["production"]["ph-a"]["p"]  | int(0) }}{%endif%}'
    state_class: measurement
    device_class: power

  - platform: mqtt
    state_topic: "/envoy/json"
    value_template: '{{ value_json["total-consumption"]["ph-a"]["p"] }}'
    name: "mqtt_consumption"
    qos: 0
    unit_of_measurement: "W"
    state_class: measurement
    device_class: power

  - platform: template
    sensors:
      exporting:
        friendly_name: "Current MQTT Energy Exporting"
        value_template: "{{ [0, (states('sensor.mqtt_production') | int(0) - states('sensor.mqtt_consumption') | int(0))] | max }}"
        unit_of_measurement: "W"
        icon_template: mdi:flash
      importing:
        friendly_name: "Current MQTT Energy Importing"
        value_template: "{{ [0, (states('sensor.mqtt_consumption') | int(0) - states('sensor.mqtt_production') | int(0))] | max }}"
        unit_of_measurement: "W"
        icon_template: mdi:flash
      solarpower:
        friendly_name: "Solar MQTT Power"
        value_template: "{{ states('sensor.mqtt_production')}}"
        unit_of_measurement: "W"
        icon_template: mdi:flash

sensor 1001:
  - platform: integration
    source: sensor.solarpower
    name: energy_solar
    method: left
#    unit: kWh
    unit_prefix: k
    unit_time: h
    round: 2
sensor 1002:
  - platform: integration
    source: sensor.exporting
    name: energy_exporting
    method: left
#    unit: kWh
    unit_prefix: k
    unit_time: h
    round: 2
sensor 1003:
  - platform: integration
    source: sensor.importing
    name: energy_importing
    method: left
#    unit: kWh
    unit_prefix: k
    unit_time: h
    round: 2 
```

sddgit, 2022-05-14:
A thought - do you have a single or 3-phase system? You’ll need to adjust the sensors if it’s 3-phase.

timolow, 2022-05-14:
I am not sure, I have standard US residential electric, is there any way I can tell by visiting the enphase dashboard?

vk2him, 2022-05-14:
It’s easy to tell if you have two or three phases - please navigate to http://your_envoy_ip/stream/meter and post back here a sample output please.

timolow, 2022-05-14:
> data: {"production":{"ph-a":{"p":1425.106,"q":267.66,"s":1455.476,"v":121.981,"i":11.932,"pf":0.98,"f":60.0},"ph-b":{"p":1412.965,"q":255.507,"s":1442.333,"v":120.886,"i":11.932,"pf":0.97,"f":60.0},"ph-c":{"p":0.0,"q":0.0,"s":0.0,"v":0.0,"i":0.0,"pf":0.0,"f":0.0}},"net-consumption":{"ph-a":{"p":-824.356,"q":-140.062,"s":894.075,"v":122.006,"i":7.33,"pf":-0.92,"f":60.0},"ph-b":{"p":893.219,"q":-439.596,"s":1125.802,"v":120.838,"i":9.28,"pf":0.79,"f":60.0},"ph-c":{"p":0.0,"q":0.0,"s":0.0,"v":0.0,"i":0.0,"pf":0.0,"f":0.0}},"total-consumption":{"ph-a":{"p":600.75,"q":-407.722,"s":561.489,"v":121.994,"i":4.603,"pf":1.0,"f":60.0},"ph-b":{"p":2306.184,"q":-695.104,"s":2563.706,"v":120.862,"i":21.212,"pf":0.9,"f":60.0},"ph-c":{"p":0.0,"q":0.0,"s":0.0,"v":0.0,"i":0.0,"pf":0.0,"f":0.0}}}

vk2him, 2022-05-14:
Thanks for that - you have a dual phase system - note you have numbers for “ph-a” and “ph-b”. This is phase a and phase b. It’s an easy fix - in the sensors you created simply modify the value template in these two lines. If you don’t know how to do that let me know and I’ll post it shortly. 

‘’’ sensor:
  - platform: mqtt
    state_topic: "/envoy/json"
    name: "mqtt_production"
    qos: 0
    unit_of_measurement: "W"
    value_template: '{% if is_state("sun.sun", "below_horizon")%}0{%else%}{{ value_json["production"]["ph-a"]["p"]  | int(0) }}{%endif%}'
    state_class: measurement
    device_class: power

  - platform: mqtt
    state_topic: "/envoy/json"
    value_template: '{{ value_json["total-consumption"]["ph-a"]["p"] }}'
    name: "mqtt_consumption"
    qos: 0
    unit_of_measurement: "W"
    state_class: measurement
    device_class: power
‘’’

timolow, 2022-05-14:
If you can post the 2 phase power update to the sensors when you have time I would appreciate it.

vk2him, 2022-05-14:
Will do shortly. Cheers

vk2him, 2022-05-14:
Before I post the updated code, can you please confirm the code you posted above is EXACTLY how it is within your configuration.yaml? I'm asking because as you've posted it, sensor: is listed twice and that won't work as each sensor: must have a unique number.
This is a truncated version of what you posted - note sensor: at the start and end - that won't work?

```
sensor:
  - platform: mqtt
    state_topic: "/envoy/json"
    name: "mqtt_production"
    qos: 0
    unit_of_measurement: "W"
    value_template: '{% if is_state("sun.sun", "below_horizon")%}0{%else%}{{ value_json["production"]["ph-a"]["p"]  | int(0) }}{%endif%}'
    state_class: measurement
    device_class: power

  - platform: mqtt
    state_topic: "/envoy/json"
    value_template: '{{ value_json["total-consumption"]["ph-a"]["p"] }}'
    name: "mqtt_consumption"
    qos: 0
    unit_of_measurement: "W"
    state_class: measurement
    device_class: power

  - platform: mqtt
    state_topic: "/envoy/json"
    name: "mqtt_power_factor"
    qos: 0
    unit_of_measurement: "%"
    value_template: '{{ value_json["total-consumption"]["ph-a"]["pf"] }}'
    state_class: measurement
    device_class: power_factor

  - platform: mqtt
    state_topic: "/envoy/json"
    name: "mqtt_voltage"
    qos: 0
    unit_of_measurement: "V"
    value_template: '{{ value_json["total-consumption"]["ph-a"]["v"] }}'
    state_class: measurement
    device_class: voltage
sensor:

timolow, 2022-05-14:
I think I was confused by the readme on the project and I copy/pasted both sensor stanzas.  Config as follows:

> sensor:
>   - platform: mqtt
>     state_topic: "/envoy/json"
>     name: "mqtt_production"
>     qos: 0
>     unit_of_measurement: "W"
>     value_template: '{% if is_state("sun.sun", [...]

vk2him, 2022-05-14:
Thanks- that’s strange as you definitely have a duplicate sensor: - this should be throwing errors in your logs. Can you confirm that sensor 999: is unused and I’ll update that entire code for you. I’ll also make a change as currently it’s ignoring the numbers as decimals so that will also give a slight variance to the online Enphase readings you’re comparing to.

timolow, 2022-05-14:
Sensor 999 is unused, Should i remove the 2nd sensor: stanza or the 2nd sensor stanza and mqtt_production, mqtt_consumption and the 3 templates?

vk2him, 2022-05-14:
Thanks Tim - just leave it for now - replace the entire section with the "fixed" version I'm just about to post

vk2him, 2022-05-14:
Replace your current code entirely with this. Note I've added three phases just in case you upgrade one day. 
Notes:
1) I also assume sensor 1000: is free?
2) I've renamed the power factor and voltage sensors you created as I've included both phase a and phase b for them - You'll have to change your dashboard etc to display these new ones (I appended _phasea or _phaseb to the sensor names)
3) Let me know how this goes?

```
sensor 999:
  - platform: mqtt
    state_topic: "/envoy/json"
    name: "mqtt_production"
    qos: 0
    unit_of_measurement: "W"
    value_template: '{% if is_state("sun.sun", "below_horizon")%}0{%else%}{{ value_json["production"]["ph-a"]["p"]  | float(0) + value_json["production"]["ph-b"]["p"]  | float(0) +value_json["production"]["ph-c"]["p"]  | float(0) }}{%endif%}'
    state_class: measurement
    device_class: power

  - platform: mqtt
    state_topic: "/envoy/json"
    value_template: '{{ value_json["total-consumption"]["ph-a"]["p"] | float(0) + value_json["total-consumption"]["ph-b"]["p"] | float(0) + value_json["total-consumption"]["ph-c"]["p"] | float(0) }}'
    name: "mqtt_consumption"
    qos: 0
    unit_of_measurement: "W"
    state_class: measurement
    device_class: power

  - platform: mqtt
    state_topic: "/envoy/json"
    name: "mqtt_power_factor_phasea"
    qos: 0
    unit_of_measurement: "%"
    value_template: '{{ value_json["total-consumption"]["ph-a"]["pf"] }}'
    state_class: measurement
    device_class: power_factor

  - platform: mqtt
    state_topic: "/envoy/json"
    name: "mqtt_power_factor_phaseb"
    qos: 0
    unit_of_measurement: "%"
    value_template: '{{ value_json["total-consumption"]["ph-b"]["pf"] }}'
    state_class: measurement
    device_class: power_factor    

  - platform: mqtt
    state_topic: "/envoy/json"
    name: "mqtt_voltage_phasea"
    qos: 0
    unit_of_measurement: "V"
    value_template: '{{ value_json["total-consumption"]["ph-a"]["v"] }}'
    state_class: measurement
    device_class: voltage

  - platform: mqtt
    state_topic: "/envoy/json"
    name: "mqtt_voltage_phaseb"
    qos: 0
    unit_of_measurement: "V"
    value_template: '{{ value_json["total-consumption"]["ph-b"]["v"] }}'
    state_class: measurement
    device_class: voltage

sensor 1000:

  #
  # These ones are for Envoy via mqtt
  #
  - platform: mqtt
    state_topic: "/envoy/json"
    name: "mqtt_production"
    qos: 0
    unit_of_measurement: "W"
    value_template: '{% if is_state("sun.sun", "below_horizon")%}0{%else%}{{ value_json["production"]["ph-a"]["p"]  | float(0) + value_json["production"]["ph-b"]["p"]  | float(0) + value_json["production"]["ph-c"]["p"]  | float(0) }}{%endif%}'
    state_class: measurement
    device_class: power

  - platform: mqtt
    state_topic: "/envoy/json"
    value_template: '{{ value_json["total-consumption"]["ph-a"]["p"] | float(0) + value_json["total-consumption"]["ph-b"]["p"] | float(0) + value_json["total-consumption"]["ph-c"]["p"] | float(0) }}'
    name: "mqtt_consumption"
    qos: 0
    unit_of_measurement: "W"
    state_class: measurement
    device_class: power

  - platform: template
    sensors:
      exporting:
        friendly_name: "Current MQTT Energy Exporting"
        value_template: "{{ [0, (states('sensor.mqtt_production') | int(0) - states('sensor.mqtt_consumption') | int(0))] | max }}"
        unit_of_measurement: "W"
        icon_template: mdi:flash
      importing:
        friendly_name: "Current MQTT Energy Importing"
        value_template: "{{ [0, (states('sensor.mqtt_consumption') | int(0) - states('sensor.mqtt_production') | int(0))] | max }}"
        unit_of_measurement: "W"
        icon_template: mdi:flash
      solarpower:
        friendly_name: "Solar MQTT Power"
        value_template: "{{ states('sensor.mqtt_production')}}"
        unit_of_measurement: "W"
        icon_template: mdi:flash

sensor 1001:
  - platform: integration
    source: sensor.solarpower
    name: energy_solar
    method: left
#    unit: kWh
    unit_prefix: k
    unit_time: h
    round: 2
sensor 1002:
  - platform: integration
    source: sensor.exporting
    name: energy_exporting
    method: left
#    unit: kWh
    unit_prefix: k
    unit_time: h
    round: 2
sensor 1003:
  - platform: integration
    source: sensor.importing
    name: energy_importing
    method: left
#    unit: kWh
    unit_prefix: k
    unit_time: h
    round: 2
```

vk2him, 2022-05-14:
Sorry Tim - I had a Sunday morning brain fade - You'll probably see that sensor 999: and sensor 1000: have mqtt_production and mqtt_consumption so I've duplicated them. The correct code is as follows so please use that instead of the above:

```
sensor 999:

  - platform: mqtt
    state_topic: "/envoy/json"
    name: "mqtt_power_factor_phasea"
    qos: 0
    unit_of_measurement: "%"
    value_template: '{{ value_json["total-consumption"]["ph-a"]["pf"] }}'
    state_class: measurement
    device_class: power_factor

  - platform: mqtt
    state_topic: "/envoy/json"
    name: "mqtt_power_factor_phaseb"
    qos: 0
    unit_of_measurement: "%"
    value_template: '{{ value_json["total-consumption"]["ph-b"]["pf"] }}'
    state_class: measurement
    device_class: power_factor    

  - platform: mqtt
    state_topic: "/envoy/json"
    name: "mqtt_voltage_phasea"
    qos: 0
    unit_of_measurement: "V"
    value_template: '{{ value_json["total-consumption"]["ph-a"]["v"] }}'
    state_class: measurement
    device_class: voltage

  - platform: mqtt
    state_topic: "/envoy/json"
    name: "mqtt_voltage_phaseb"
    qos: 0
    unit_of_measurement: "V"
    value_template: '{{ value_json["total-consumption"]["ph-b"]["v"] }}'
    state_class: measurement
    device_class: voltage

sensor 1000:

  #
  # These ones are for Envoy via mqtt
  #
  - platform: mqtt
    state_topic: "/envoy/json"
    name: "mqtt_production"
    qos: 0
    unit_of_measurement: "W"
    value_template: '{% if is_state("sun.sun", "below_horizon")%}0{%else%}{{ value_json["production"]["ph-a"]["p"]  | float(0) + value_json["production"]["ph-b"]["p"]  | float(0) + value_json["production"]["ph-c"]["p"]  | float(0) }}{%endif%}'
    state_class: measurement
    device_class: power

  - platform: mqtt
    state_topic: "/envoy/json"
    value_template: '{{ value_json["total-consumption"]["ph-a"]["p"] | float(0) + value_json["total-consumption"]["ph-b"]["p"] | float(0) + value_json["total-consumption"]["ph-c"]["p"] | float(0) }}'
    name: "mqtt_consumption"
    qos: 0
    unit_of_measurement: "W"
    state_class: measurement
    device_class: power

  - platform: template
    sensors:
      exporting:
        friendly_name: "Current MQTT Energy Exporting"
        value_template: "{{ [0, (states('sensor.mqtt_production') | int(0) - states('sensor.mqtt_consumption') | int(0))] | max }}"
        unit_of_measurement: "W"
        icon_template: mdi:flash
      importing:
        friendly_name: "Current MQTT Energy Importing"
        value_template: "{{ [0, (states('sensor.mqtt_consumption') | int(0) - states('sensor.mqtt_production') | int(0))] | max }}"
        unit_of_measurement: "W"
        icon_template: mdi:flash
      solarpower:
        friendly_name: "Solar MQTT Power"
        value_template: "{{ states('sensor.mqtt_production')}}"
        unit_of_measurement: "W"
        icon_template: mdi:flash

sensor 1001:
  - platform: integration
    source: sensor.solarpower
    name: energy_solar
    method: left
#    unit: kWh
    unit_prefix: k
    unit_time: h
    round: 2
sensor 1002:
  - platform: integration
    source: sensor.exporting
    name: energy_exporting
    method: left
#    unit: kWh
    unit_prefix: k
    unit_time: h
    round: 2
sensor 1003:
  - platform: integration
    source: sensor.importing
    name: energy_importing
    method: left
#    unit: kWh
    unit_prefix: k
    unit_time: h
    round: 2

```

vk2him, 2022-05-14:
Thinking about this - since mqtt_production and mqtt_consumption were duplicated in your original configuration using the same sensor: identifier, it would have still worked as the second sensor: may have over-ridden the first sensor: - you probably wouldn't have been able to see power factor or voltage I'd guess with your original config?

timolow, 2022-05-15:
I have replaced/added this to my config and it seems to be working as intended, I created a power-wheel-card and the realtime consumption is on par with enphase monitoring now.

vk2him, 2022-05-15:
Ok great thanks

closed 2022-05-15 by vk2him

sddgit, 2022-05-15:
@vk2him Do you have any thoughts on using this approach to integrate with Energy vs the older approach at https://community.home-assistant.io/t/enhpase-envoy-on-2021-8-with-new-energy-feature/328668? That will update at the frequency of the Enphase integration (every minute, I’m not sure?) vs this approach of every second. Given the Energy component gives pretty coarse readings, I’m wondering whether updating it every second produces any unnecessary overhead. Thanks.

vk2him, 2022-05-15:
Yes you’re right that the per second updates are overkill for energy- the only impact on resource would be disk space I’d guess however I haven’t experienced any issues.

sddgit, 2022-05-15:
Yes, I suppose every update to the sensors would be recorded, and have to be for the Energy history.