vk2him/Enphase-Envoy-mqtt-json #9: Add on fails to start: s6-overlay-suexec: fatal: can only run as pid 1

rct, 2022-05-15:
Trying to run this as an add on on a 2022.5.4 system generates the error: `s6-overlay-suexec: fatal: can only run as pid 1`

Something seems to have changed w.r.t hass add on configurations.  Active discussion is currently occurring in this community thread:

[Hello_world example addon from developer docs stopped working (s6 overlay issue?) ](https://community.home-assistant.io/t/hello-world-example-addon-from-developer-docs-stopped-working-s6-overlay-issue/421486)

Adding `init: false` to the add on's `config.yaml` might fix this, though there seems to be more discussion about updating base images.

vk2him, 2022-05-15:
It’s running fine on my 2022.5.4 system running Home Assistant OS7.6- are you running HA in a container or Home Assistant OS?

vk2him, 2022-05-16:
@rct - thanks for that - I've added `init: false` per your commit - it is working on my system without error however I didn't have an error previously. Strange that some system have it.

rct, 2022-05-16:
@vk2him - Yes, it is working for me since adding `init: false`.  Not sure why it isn't tripping you up. Maybe you haven't rebuilt with the v3 s6 base/components yet?  

[S6-Overlay 3.x update on our docker base images](
https://developers.home-assistant.io/blog/2022/05/12/s6-overlay-base-images/) from Home Assistant Developer blog has some more info.

----

Yes I've made the MQTT_TOPIC configurable.  Also, I've removed the leading `/` from the topic.  I'm still somewhat new to MQTT, but I believe that's creating a topic path that is `(null)/envoy/json`.  It could be seen in the MQTT explorer GUI as having an empty top level topic string.   Of course that change will break existing `state_topic` definitions for sensors.

Thanks for creating this project. I've been putting off trying to create my own addons, but I've been learning from your project, so thanks for that.

One more thing - the code as currently written won't work standalone unless you have a `/data/options.json` file.  So there needs to be a way to set where that is or try a few paths.

vk2him, 2022-05-16:
@rct - yes you're right - I didn't need to rebuild so that's probably why I didn't see it - thanks for raising the issue and providing the fix!

Re using it standalone - I mentioned in the Readme under Installation Method 2 that  `/data/options.json` is needed - as an alternative you could also hard-code the values into '[envoy_to_mqtt_json.py' - this is how I originally used it. 

Cheers

closed 2022-05-16 by vk2him