vk2him/Enphase-Envoy-mqtt-json #11: Install in Docker with docker-compose (request)

cmseasy, 2022-11-08:
I am running Home Assistant Container in Docker, not in HA OS. I have no Add-On's
Is there a workout (script?) to install Enphase-Envoy-mqtt-json] in docker (with docker-compose)?

vk2him, 2022-11-09:
I'm unable to test docker, my HA is running as a Virtual Machine - I did get the script to work fine and send the correct data to the mqqt broker running as a standalone application from the Linux OS that hosts the VM. Checkout Installation Method 2 from my instructions as you may be able to run it that way.

vk2him, 2022-11-09:
Thinking about this more, there may be a way add this manually within HA.
 
Create a directory from root called  /addons/Enphase-Envoy-mqtt-json then copy all the files manually from my github repo into it – you will have something like this: - Note the new directory is on the same level as  the /config directory:
 
 
[image]

 
Restart HA and you should see it in the Add-on Store as a Local add-ons
 
[image]
 
Click that to install per the Instructions.

Some help might be available here on this as that is what I followed to create the add-on initially:
 
https://developers.home-assistant.io/docs/add-ons/tutorial/
 
Cheers
Ian

cmseasy, 2022-11-09:
Thanks for your reaction/help.

Unfortunately addons are not added in a Docker Home Assistant Container envoironment. Addons needs the not installed Supervisor. I tested your workflow: no addons in Docker HA.

I read that Addons are Docker containers, so it shoult be possible to install  Enphase-Envoy-mqtt-json in a container. My first Enphase-Envoy-mqtt-json Docker install finished with an error. I wil try further, I am not a Docker expert... it is a learning proces...

Once again: thanks,
Martin

closed 2023-03-27 by vk2him

DMedina559, 2024-03-13:
Hi , @cmseasy sorry for responding to this old thread, but did you ever figure this out?

cmseasy, 2024-03-13:
> did you ever figure this out?

No, I did not. 
I installed HAOS with Supervisor in a Proxmox virtual machine on a NUC and moved my configuration.