vk2him/Enphase-Envoy-mqtt-json #14: Getting other data from enphase

FoliniC, 2023-02-16:
Great! I'm using it in my ha...
I'm also interested in getting other data that comes from enphase. In particular panels data.
Did you plan to publish those data in the future?
Data that came from my endpoints (taken in the evening so no production from my 17 panels):

**http://<server>/home.json**
{
...."alerts":.[
....],
...."comm":.{
........"acb":.{
............"level":.0,
............"num":.0
........},
........"level":.0,
........"nsrb":.{
............"level":.5,
............"num":.1
........},
........"num":.1,
........"pcu":.{
............"level":.0,
............"num":.0
........}
....},
...."current_date":."02/16/2023",
...."current_time":."21:43",
...."db_percent_full":."5",
...."db_size":."20.MB",
...."is_nonvoy":.false,
...."network":.{
........"ever_reported_to_enlighten":.true,
........"interfaces":.[
............{
................"carrier":.true,
................"dhcp":.false,
................"interface":."eth0",
................"ip":."192.168.0.2",
................"mac":."00:11:11:11:11:11",
................"signal_strength":.1,
................"signal_strength_max":.1,
................"type":."ethernet"
............},
............{
................"carrier":.false,
................"configured":.true,
................"dhcp":.true,
................"interface":."wlan0",
................"ip":.null,
................"mac":."48:11:11:11:11:11",
................"present":.true,
................"signal_strength":.0,
................"signal_strength_max":.0,
................"status":."connecting",
................"supported":.true,
................"type":."wifi"
............}
........],
........"last_enlighten_report_time":.1676579382,
........"primary_interface":."eth0",
........"web_comm":.true
....},
...."software_build_epoch":.1622651920,
...."tariff":."none",
...."timezone":."Europe/Rome",
...."update_status":."not-satisfied"
}
**http://<Server>/ivp/meters**

[
[... 487 lines cut ...]
................"device_status":.[
...................."envoy.global.ok"
................],
................"img_load_date":."1625075612",
................"img_pnum_running":."520-01794-r01-v01.02.19",
................"installed":."1666614425",
................"last_rpt_date":."1676580366",
................"line1-connected":.true,
................"line2-connected":.false,
................"line3-connected":.false,
................"line-count":.3,
................"operating":.true,
................"part_num":."800-00606-r02",
................"producing":.false,
................"provisioned":.true,
................"ptpn":."540-01794-r01-v01.02.19",
................"reason":."ok",
................"reason_code":.-1,
................"relay":."closed",
................"serial_num":."122139011111"
............}
........],
........"type":."NSRB"
....}
]

vk2him, 2023-02-16:
The built-in Enphase integration provides per panel data if you use the installer username/password when you set it up. It works well, I can see all 18 of my panels with this. https://www.home-assistant.io/integrations/enphase_envoy/

FoliniC, 2023-02-17:
I know that it's possible from the integration, but you get only data every few minutes. I want to check every few seconds. Doing this could reveal "peak" anomaly (like the one I discovered on Q-Relay)

vk2him, 2023-02-17:
I'm not aware of an endpoint that displays per panel data any quicker than once per 15 minutes - The one I know is this using the installer username. http://your-envoy-ip/api/v1/production/inverters
I believe that url is what the inbuilt integration uses.
This page shows the other endpoints available, there's no other for individual inverters.
https://thecomputerperson.wordpress.com/2016/08/03/enphase-envoy-s-data-scraping/

closed 2023-03-27 by vk2him