vk2him/Enphase-Envoy-mqtt-json #15: Any plans to update this to support token auth?

enkrypt3d, 2023-03-27:
It'd be great to be able to use this for v7 and pull the real time data from envoy.local/ivp/meters/readings ?? Any updates planned? thanks

vk2him, 2023-03-27:
I saw your question on the HA forum - short answer is no.
Refer to this answer I posed the other day https://github.com/vk2him/Enphase-Envoy-mqtt-json/issues/5#issuecomment-1482418068
Cheers

closed 2023-03-27 by vk2him

Matthew1471, 2023-06-05:
I've written a simple library for V7 that you could consume if that would help? I'm going to add a AMQP client soon as other apps in my house would benefit from it.

Example: https://github.com/Matthew1471/Enphase-API/blob/main/Python/examples/gateway_database_meters.py

API Documentation: https://github.com/Matthew1471/Enphase-API/blob/main/Documentation/IQ%20Gateway%20API/Stream/Meter.adoc

Turbox35, 2023-06-20:
Hello Matthew1471,
It seems you know exactly how to generates the token and use it. Are you able/interested to make a fork from this repo and add the missing token?

Matthew1471, 2023-06-27:
Hi @Turbox35 , I prefer AMQP personally (which some brokers do support MQTT - so there is some interopability). I've been wanting to make the switch to a queue on my project, so there's a implementation of AMQP broker support here : https://github.com/Matthew1471/Enphase-API/blob/main/Python/examples/gateway_amqp_meters.py

As I only have a V7 gateway to run against this natively supports the token authentication.

Turbox35, 2023-06-27:
I understand, from my side I don't know AMQP and I need a link with Domoticz and Home Assistant to my enphase gateway which can be done with MQTT discovery. Nevertheless, thank you for sharing.

Matthew1471, 2023-06-27:
It looks like it can be done with RabbitMQ's "MQTT plugin", https://www.domoticz.com/forum/viewtopic.php?t=11765

AMQP is just MQTT with extra features (security, message headers) and AMQP has a broker which is just a server whereas MQTT is generally for IoT devices that don't have much battery or storage power. https://www.educba.com/amqp-vs-mqtt/ and https://www.geeksforgeeks.org/difference-between-rabbitmq-vs-mqtt/

I don't know anything about Domoticz and little about Home Assistant.. but what "server" do you use for the MQTT parts?

vk2him, 2023-07-19:
@enkrypt3d - with help from another github author, this addon now support v7 and provides 1 second updates. I have a beta version you can test - I'll update my main version soon. To be done - cache the v7 token so it will work when offline. You can read about it in this issue (at bottom)  https://github.com/vk2him/Enphase-Envoy-mqtt-json/issues/5
Link to the beta code - https://github.com/vk2him/Enphase-Envoy-mqtt-json-v7
Click the blue Add Repository button from the readme and follow the instructions

Turbox35, 2023-07-19:
Hello,

Super! I will try it as soon as possible but regarding the update rate, 1 second is too high for me. Can I change this parameter? I have 3 systems which already poll my envoy.

vk2him, 2023-07-19:
@Turbox35 This integration is basically designed for 1 second updates for devices that require "instant" response or to display real-time graphs - the HA native Enphase integration does 30 second updates and may be better suited to your needs?

Turbox35, 2023-07-19:
I see but my needs are a little bit different.
In parallel I use Home Assistant/Domoticz and Grafana. That's why I prefer an autonomous system which provide data to MQTT (with autodiscovey for easy implementation).
Some users have report unresponsive Envoy if the pollind rate is too high. I prefer to start slowly (30sec in a first time) and if all works well as expected I will removed the integrated automation plugins and then increase the polling rate frequency (why not 1second). My wish is to use a standalone docker.

vk2him, 2023-07-19:
It's a stream, so no polling - it connects once to the envoy's stream and then digests that - very low impact on the envoy compared to polling it every second. The people complaining about 1 second issues on the Enphase Community forum are not using the stream, they are using the non stream end points.

del13r, 2023-07-19:
I have a rest solution where you can choose the update interval yourself.
https://community.home-assistant.io/t/enphase-envoy-with-energy-dashboard/328668/690?u=del13r

Turbox35, 2023-07-19:
Should I feed "config.yaml" or "data/option.json"?

Turbox35, 2023-07-19:
I have proposed modifications. It starts but now I have an issue with mqtt.
`19/07/2023 23:07:33 Detected FW version 7
Traceback (most recent call last):
  File "/home/turbox/envoy/Enphase-Envoy-mqtt-json-v7/envoy_to_mqtt_json.py", line 185, in <module>
    client.connect(MQTT_HOST,int(MQTT_PORT), 30)
  File "/usr/local/lib/python3.11/dist-packages/paho_mqtt-1.6.1-py3.11.egg/paho/mqtt/client.py", line 914, in connect
    return self.reconnect()
           ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/paho_mqtt-1.6.1-py3.11.egg/paho/mqtt/client.py", line 1045, in reconnect
    on_pre_connect = self.on_pre_connect
                     ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/paho_mqtt-1.6.1-py3.11.egg/paho/mqtt/client.py", line 1863, in on_pre_connect
    return self._on_pre_connect
           ^^^^^^^^^^^^^^^^^^^^
AttributeError: 'Client' object has no attribute '_on_pre_connect'. Did you mean: 'on_pre_connect'?
`

vk2him, 2023-07-19:
Can you paste the contents of the addon Configuration page to help troubleshoot please? Be sure your mqtt and envoy user passwords are correct as well as your envoy user id. This is mine:

[image]

Turbox35, 2023-07-20:
I don't use add-on integration into HA but manual execution in Debian virtual machine.
Without the parameters proposed and removing the "/" of loading file "/data/options.json" the script doesn't start.

vk2him, 2023-07-20:
If you're using it standalone, then you must enter the correct username/passwords etc into /data/options.json  ie, there values need to be set to correct values and not left as the default. It will work standalone, thats how I initially used it for 6 months of so before I created the HA addon.
```
{
    "MQTT_HOST": "ip.address.of.homeasistant",
    "MQTT_USER": "mqtt.user",
    "MQTT_PASSWORD": "mqtt.password",
    "MQTT_PORT": "1883",
    "MQTT_TOPIC": "envoy/data",
    "ENVOY_HOST": "envoy.local",
    "ENVOY_USER": "envoy.user",
    "ENVOY_USER_PASS": "envoy.user.password",
    "ENVOY_PASSWORD": "envoy.password",
    "TOPIC_FREEDS": "Inverter/GridWatts",
    "ENVOY_TOKEN": "generate token in https://entrez.enphaseenergy.com/entrez_tokens"
  }```

Turbox35, 2023-07-20:
I have tried the last version of V7, now I have 'data/token.txt" No such file or directory