vk2him/Enphase-Envoy-mqtt-json #16: Seeing errors with enphase integration

enkrypt3d, 2023-04-06:
2023-04-06 11:19:34.731 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.envoy_#########_current_battery_capacity (<class 'custom_components.enphase_envoy.sensor.TotalBatteryCapacityEntity'>) is using state class 'measurement' which is impossible considering device class ('energy') it is using; expected None or one of 'total_increasing', 'total'; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.

enkrypt3d, 2023-04-16:
Any ideas? Rebooting the envoy seems to "fix" it for a while but over time the issue comes back.

vk2him, 2023-04-16:
My integration doesn't supply battery data unless you've modified it to pickup a different data feed? What sensor is feeding this battery entity and where does that data come from?

enkrypt3d, 2023-04-16:
Sorry I commented on the wrong integration! I Meant to put my comment on this:

https://github.com/whisk3y/home_assistant_envoy_d7_fw/tree/brianfork

closed 2023-04-16 by enkrypt3d