vk2him/Enphase-Envoy-mqtt-json #18: Error

edwin123456, 2023-07-21:
I get this error

21/07/2023 07:14:56 Connected to 192.168.2.130:1883
21/07/2023 07:14:56 Subscribed to MQTT_TOPIC: /envoy/json
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "envoy_to_mqtt_json.py", line 353, in scrape_stream_meters
    stream = requests.get(url, timeout=5, verify=False, headers=headers)
  File "/home/pi/.local/lib/python3.7/site-packages/requests/api.py", line 75, in get
    return request('get', url, params=params, **kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/requests/api.py", line 60, in request
    return session.request(method=method, url=url, **kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/requests/sessions.py", line 533, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/requests/sessions.py", line 646, in send
    r = adapter.send(request, **kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/requests/adapters.py", line 449, in send
    timeout=timeout
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 600, in urlopen
    chunked=chunked)
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 354, in _make_request
    conn.request(method, url, **httplib_request_kw)
  File "/usr/lib/python3.7/http/client.py", line 1260, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/usr/lib/python3.7/http/client.py", line 1301, in _send_request
    self.putheader(hdr, value)
  File "/usr/lib/python3.7/http/client.py", line 1238, in putheader
    raise ValueError('Invalid header value %r' % (values[i],))

What could be the problem?

vk2him, 2023-07-21:
Hi, do you have v5 or v7 envoy firmware? The error is from the v7 section so would be good if you can confirm. Are you new to using this or has it worked on previous versions?

edwin123456, 2023-07-21:
I have v7 version
And new to this,
At first the script couldn’t find the json file so i added the right path in the sript , then it couldn’t find the text file I added that path as well
Then it worked a bit thill the error. 

thanks for helping me. 
Sorry for my English

vk2him, 2023-07-21:
Ok thanks - it's not clear, are you running this in Home Assistant or stand alone? When you mentioned it couldn't find the json file or the text file, it sounds like you're running it standalone because these are usually set in the configuration page in Home Assistant.  If you're running it standalone, have a read towards the bottom of the following issue as another user is running stand alone and he posted some tips https://github.com/vk2him/Enphase-Envoy-mqtt-json/issues/5

vk2him, 2023-07-27:
@edwin123456 - any update please? Thanks

edwin123456, 2023-07-27:
I have version v7
And using it stand alone
The tip from the other user didnt help. 
error stays the same

Thanks

vk2him, 2023-07-28:
Hi, can you please try updating as some debug code has been added to hopefully identify where the issues may be. Turn on the Debug switch in Configuration and copy/paste the log here please after you remove your long token. Then turn off debug as it can fill the log quickly. Thanks

edwin123456, 2023-07-28:
Thanks for your help
Sorry Im a noob at this .
I got:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 159, in _new_conn
    (self._dns_host, self.port), self.timeout, **extra_kw)
  File "/usr/lib/python3/dist-packages/urllib3/util/connection.py", line 80, in create_connection
    raise err
  File "/usr/lib/python3/dist-packages/urllib3/util/connection.py", line 70, in create_connection
    sock.connect(sa)
OSError: [Errno 113] No route to host

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 600, in urlopen
    chunked=chunked)
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 354, in _make_request
    conn.request(method, url, **httplib_request_kw)
  File "/usr/lib/python3.7/http/client.py", line 1260, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/usr/lib/python3.7/http/client.py", line 1306, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.7/http/client.py", line 1255, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.7/http/client.py", line 1030, in _send_output
    self.send(msg)
  File "/usr/lib/python3.7/http/client.py", line 970, in send
    self.connect()
  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 181, in connect
    conn = self._new_conn()
  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 168, in _new_conn
    self, "Failed to establish a new connection: %s" % e)
urllib3.exceptions.NewConnectionError: <urllib3.connection.HTTPConnection object at 0xb59a4b50>: Failed to establish a new connection: [Errno 113] No route to host

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/requests/adapters.py", line 449, in send
    timeout=timeout
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 638, in urlopen
    _stacktrace=sys.exc_info()[2])
  File "/usr/lib/python3/dist-packages/urllib3/util/retry.py", line 398, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPConnectionPool(host='192.168.2.199', port=80): Max retries exceeded with url: /info (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xb59a4b50>: Failed to establish a new connection: [Errno 113] No route to host'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "envoy_to_mqtt_json.py", line 82, in <module>
    response_info = requests.get(url_info, verify=False)
  File "/home/pi/.local/lib/python3.7/site-packages/requests/api.py", line 75, in get
    return request('get', url, params=params, **kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/requests/api.py", line 60, in request
    return session.request(method=method, url=url, **kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/requests/sessions.py", line 533, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/requests/sessions.py", line 646, in send
    r = adapter.send(request, **kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/requests/adapters.py", line 516, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPConnectionPool(host='192.168.2.199', port=80): Max retries exceeded with url: /info (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xb59a4b50>: Failed to establish a new connection: [Errno 113] No route to host'))

Thanks

vk2him, 2023-07-28:
If you're running this standalone, then you have to enter all configuration parameters manually in the file data/options.json  - have you done that? The error is strange, it's trying to connect via http port 80 however for a V7 system it should be https port 443 . The final error No route to host may give some clues - is the system you're running this on the same network as your envoy, ie, 192.168.2.xxx?

edwin123456, 2023-07-28:
I found out the ip adress had changed, now i got this error:

28/07/2023 08:37:03 Serial number: 122223030470
28/07/2023 08:37:03 Detected FW version 7
28/07/2023 08:37:03 FREEDS is active, using topic: Inverter/GridWatts
28/07/2023 08:37:03 Read token from file data/token.txt :  

28/07/2023 08:37:03 Will wait for mqtt connect
28/07/2023 08:37:03 Finished waiting for mqtt connect
28/07/2023 08:37:03 Connected to 192.168.2.130:1883
28/07/2023 08:37:03 Url: https://192.168.2.194/ivp/meters/readings
28/07/2023 08:37:03 headers: {'Authorization': 'Bearer \n'}
28/07/2023 08:37:03 Subscribed to MQTT_TOPIC: /envoy/json
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "envoy_to_mqtt_json.py", line 358, in scrape_stream_meters
    stream = requests.get(url, timeout=5, verify=False, headers=headers)
  File "/home/pi/.local/lib/python3.7/site-packages/requests/api.py", line 75, in get
    return request('get', url, params=params, **kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/requests/api.py", line 60, in request
    return session.request(method=method, url=url, **kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/requests/sessions.py", line 533, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/requests/sessions.py", line 646, in send
    r = adapter.send(request, **kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/requests/adapters.py", line 449, in send
    timeout=timeout
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 600, in urlopen
    chunked=chunked)
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 354, in _make_request
    conn.request(method, url, **httplib_request_kw)
  File "/usr/lib/python3.7/http/client.py", line 1260, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/usr/lib/python3.7/http/client.py", line 1301, in _send_request
    self.putheader(hdr, value)
  File "/usr/lib/python3.7/http/client.py", line 1238, in putheader
    raise ValueError('Invalid header value %r' % (values[i],))
ValueError: Invalid header value b'Bearer \n'

vk2him, 2023-07-28:
On this line of the log, have you truncated/redacted/edited the output before posting as it should be displaying the very long token?

``` 28/07/2023 08:37:03 Read token from file data/token.txt :  ```

Similarly, this line should have printed the same very long token - did you remove it before posing here?

``` 28/07/2023 08:37:03 headers: {'Authorization': 'Bearer \n'}```

If you didn't remove it, then what is actually contained in the file ```data/token.txt``` as it should be the valid 400+ APK Token?

edwin123456, 2023-07-28:
I didnt remove it 
I looked at the text file and it was empty then i got a new token en copied it in the text file but then i got an error that its the wrong token?

Verstuurd vanaf mijn iPhone

vk2him, 2023-07-28:
No need to enter anything into data/token.txt - the script will do it for you. Try deleting data/token.txt then restart the script

edwin123456, 2023-07-28:
And when i copy and paste the token from the text file
In the envoy website it works
So the token is good
Strange problem

Thanks for your help

vk2him, 2023-07-28:
What is the error in log now?

edwin123456, 2023-07-28:
I erased the token and this is the error:

python3 envoy_to_mqtt_json.py
28/07/2023 09:59:55 Serial number: 122223030470
28/07/2023 09:59:55 Detected FW version 7
28/07/2023 09:59:55 FREEDS is active, using topic: Inverter/GridWatts
28/07/2023 09:59:55 Read token from file data/token.txt :  

28/07/2023 09:59:55 Will wait for mqtt connect
28/07/2023 09:59:55 Finished waiting for mqtt connect
28/07/2023 09:59:55 Connected to 192.168.2.130:1883
28/07/2023 09:59:55 Url: https://192.168.2.194/ivp/meters/readings
28/07/2023 09:59:55 Subscribed to MQTT_TOPIC: /envoy/json
28/07/2023 09:59:55 headers: {'Authorization': 'Bearer \n'}
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "envoy_to_mqtt_json.py", line 358, in scrape_stream_meters
    stream = requests.get(url, timeout=5, verify=False, headers=headers)
  File "/home/pi/.local/lib/python3.7/site-packages/requests/api.py", line 75, in get
    return request('get', url, params=params, **kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/requests/api.py", line 60, in request
    return session.request(method=method, url=url, **kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/requests/sessions.py", line 533, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/requests/sessions.py", line 646, in send
    r = adapter.send(request, **kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/requests/adapters.py", line 449, in send
    timeout=timeout
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 600, in urlopen
    chunked=chunked)
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 354, in _make_request
    conn.request(method, url, **httplib_request_kw)
  File "/usr/lib/python3.7/http/client.py", line 1260, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/usr/lib/python3.7/http/client.py", line 1301, in _send_request
    self.putheader(hdr, value)
  File "/usr/lib/python3.7/http/client.py", line 1238, in putheader
    raise ValueError('Invalid header value %r' % (values[i],))
ValueError: Invalid header value b'Bearer \n'

vk2him, 2023-07-28:
Hi - as a test please delete data/token.txt - I'm hoping that might show different detail. Now it's not writing the token to your file, so before you delete it, what are the permissions for data/token.txt - is it writable to the user that's running the script?

edwin123456, 2023-07-28:
I now have owner and user pi:pi
And have the same error 
Even if i run the script with sudo. 

28/07/2023 10:20:49 Serial number: 122223030470
28/07/2023 10:20:49 Detected FW version 7
28/07/2023 10:20:49 FREEDS is active, using topic: Inverter/GridWatts
28/07/2023 10:20:49 Read token from file data/token.txt :  

28/07/2023 10:20:49 Will wait for mqtt connect
28/07/2023 10:20:49 Finished waiting for mqtt connect
28/07/2023 10:20:49 Connected to 192.168.2.130:1883
28/07/2023 10:20:49 Url: https://192.168.2.194/ivp/meters/readings
28/07/2023 10:20:49 Subscribed to MQTT_TOPIC: /envoy/json
28/07/2023 10:20:49 headers: {'Authorization': 'Bearer \n'}
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "envoy_to_mqtt_json.py", line 358, in scrape_stream_meters
    stream = requests.get(url, timeout=5, verify=False, headers=headers)
  File "/home/pi/.local/lib/python3.7/site-packages/requests/api.py", line 75, in get
    return request('get', url, params=params, **kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/requests/api.py", line 60, in request
    return session.request(method=method, url=url, **kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/requests/sessions.py", line 533, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/requests/sessions.py", line 646, in send
    r = adapter.send(request, **kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/requests/adapters.py", line 449, in send
    timeout=timeout
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 600, in urlopen
    chunked=chunked)
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 354, in _make_request
    conn.request(method, url, **httplib_request_kw)
  File "/usr/lib/python3.7/http/client.py", line 1260, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/usr/lib/python3.7/http/client.py", line 1301, in _send_request
    self.putheader(hdr, value)
  File "/usr/lib/python3.7/http/client.py", line 1238, in putheader
    raise ValueError('Invalid header value %r' % (values[i],))
ValueError: Invalid header value b'Bearer \n

edwin123456, 2023-07-28:
As update:
Ls -l gives 
-rw-r--r-- 1 pi   pi     1 jul 28 09:59 token.txt

vk2him, 2023-07-28:
Have you tried deleting it and retrying

vk2him, 2023-07-28:
You're also missing options.json - that needs to be in the data directory ... this is mine

ls -l
-rw-------    1 root     root           285 Jul 28 16:33 options.json
-rw-r--r--    1 root     root           408 Jul 20 21:25 token.txt

edwin123456, 2023-07-28:
No files are missing have all with the right permissions 
I deleted the text file and run the script then i got:

Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "envoy_to_mqtt_json.py", line 356, in scrape_stream_meters
    headers = {"Authorization": "Bearer " + ENVOY_TOKEN}
TypeError: can only concatenate str (not "NoneType") to str

vk2him, 2023-07-28:
What does ls-l show now for token.txt - and is your token now in that file with nothing else

edwin123456, 2023-07-28:
-rw-r--r-- 1 pi pi 362 jul 28 10:39 options.json
-rw-r--r-- 1 pi pi   0 jul 28 10:40 token.txt

token.txt is still empty

vk2him, 2023-07-28:
ok - can you put your token into token.txt  - it should be the token only, nothing else - as a hint here is a short version of what's in mine (start and end). 

```eyJraWQiOiI3ZDEwMDA1ZC03    cut   RMfo3drAdnEQ```

edwin123456, 2023-07-28:
Did what u asked and got this error:

28/07/2023 11:02:16 Serial number: 122223030470
28/07/2023 11:02:16 Detected FW version 7
28/07/2023 11:02:16 FREEDS is active, using topic: Inverter/GridWatts
28/07/2023 11:02:16 Read token from file data/token.txt :  xxxxxxxxxDEwMDA1ZC03OXxxxxxxxxxYmNiNC0XxxxxxxxxxxxxxxxThlZTE1NmIiLCJ0eXAiOiJKV1QiLCJhbGcXxxxx

28/07/2023 11:02:16 Will wait for mqtt connect
28/07/2023 11:02:16 Finished waiting for mqtt connect
28/07/2023 11:02:16 Connected to 192.168.2.130:1883
28/07/2023 11:02:16 Url: https://192.168.2.194/ivp/meters/readings
28/07/2023 11:02:16 Subscribed to MQTT_TOPIC: /envoy/json
28/07/2023 11:02:16 headers: {'Authorization': 'Bearer xxxxxxxiI3ZDEwMDA1ZC0xxxxxxxxTRkMGQtYmNiNC0yNXxxxx\n'}
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "envoy_to_mqtt_json.py", line 358, in scrape_stream_meters
    stream = requests.get(url, timeout=5, verify=False, headers=headers)
  File "/home/pi/.local/lib/python3.7/site-packages/requests/api.py", line 75, in get
    return request('get', url, params=params, **kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/requests/api.py", line 60, in request
    return session.request(method=method, url=url, **kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/requests/sessions.py", line 533, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/requests/sessions.py", line 646, in send
    r = adapter.send(request, **kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/requests/adapters.py", line 449, in send
    timeout=timeout
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 600, in urlopen
    chunked=chunked)
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 354, in _make_request
    conn.request(method, url, **httplib_request_kw)
  File "/usr/lib/python3.7/http/client.py", line 1260, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/usr/lib/python3.7/http/client.py", line 1301, in _send_request
    self.putheader(hdr, value)
  File "/usr/lib/python3.7/http/client.py", line 1238, in putheader
    raise ValueError('Invalid header value %r' % (values[i],))
ValueError: Invalid header value b'Bearer eyJraXxxxxxxxxxxxxxxxxxQtYmNiNC0XxxxxxxxxxTE1NmIiLCJ0eXAiOiJKV1QiLCJhbGciOiJFXxxxxxxxx\n'

vk2him, 2023-07-28:
Can you look at the contents of token.txt - it appears to have this at the end which may be a new line character - ```\n```
There should not be a new line at the end, it has to be the token only. Maybe dump it in hex to verify?

vk2him, 2023-07-28:
I'm fairly certain the new line at the end is the problem you're seeing - I have a v5 system but I can "force" a token to be generated - when I do you can see from my log that there isn't a blank line:

```
28/07/2023 19:37:26 Read token from file data/token.txt :  eyJra   cut  StXK2WrenrlYpNSHMc_XA
28/07/2023 19:37:26 Connected to 192.168.1.74:1883
```

However yours displays a blank line:

```
28/07/2023 11:02:16 Read token from file data/token.txt :  xxxxxxxxxDEwMDA1ZC03OXxxxxxxxxxYmNiNC0XxxxxxxxxxxxxxxxThlZTE1NmIiLCJ0eXAiOiJKV1QiLCJhbGcXxxxx

28/07/2023 11:02:16 Will wait for mqtt connect
28/07/2023 11:02:16 Finished waiting for mqtt connect
28/07/2023 11:02:16 Connected to 192.168.2.130:1883
```

edwin123456, 2023-07-28:
How i dump it in Hex?
Sorry maybe a noob question

vk2him, 2023-07-28:
To dump in hex, do this:

```od -x token.txt```

It will output the file in hex - I suspect the last 4 digits will be 0a65 which is ```\n```

If I'm correct, what did you use to edit the token.txt file?

edwin123456, 2023-07-28:
This is my last line:

0000120 4762 6963 694f 464a 7a55 3149 694e 394a
0000140 000a
0000141

vk2him, 2023-07-28:
P.S. - Your last two hex digits are 000a - this is a linefeed so definitely wrong - it displays as \n which can also denote new line

edwin123456, 2023-07-28:
Yes, now its working
Thanks for all your help

closed 2023-07-28 by edwin123456