vk2him/Enphase-Envoy-mqtt-json #19: Doesn’t work on Envoy 5.x.x firmware

alexeiw123, 2023-07-23:
1.0.7 - works with my envoy on Software Version R5.0.55
MQTT values viewable at topic `/envoy/json`

1.0.11 - no topic updates. have tried updating envoy ip as this didn't carry through the update. There is no configuration field to provide the envoy password as there is in v1.0.7.

Perhaps I'm missing something in config but it seems this update has breaking changes, at least for my configuration.

Restoring add-on backup prior to update gets it working again.

vk2him, 2023-07-23:
Can you confirm you had /envoy/json set in configuration? What was shown in log and did you try using mqttexplorer to see if there was any output? My envoy is v5 and I didn't need to enter my Enphase password to get it to work since that's only needed for v7. 1.0.11 should work on v5 without changes needed, it's all ok on my system.

alexeiw123, 2023-07-23:
Yes, definitely have envoy/json in config.

This morning when I checked the logs it was simply a few lines about services being successfully started.

Have just upgraded again and now have this in the logs after setting my config as follows:

```
MQTT_HOST: 192.168.1.xx
MQTT_USER: <nope>
MQTT_PASSWORD: <nope>
MQTT_PORT: 1883
MQTT_TOPIC: /envoy/json
ENVOY_HOST: 192.168.1.xx
ENVOY_USER: enter.enphase.userid@email.com
ENVOY_USER_PASS: enter.enphase.password
USE_FREEDS: false
```
logs:
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
23/07/2023 16:09:31 Serial number: 122014055180
23/07/2023 16:09:31 Cannot match firmware version, got  ['R5.0.55']
23/07/2023 16:09:31 FREEDS is inactive
23/07/2023 16:09:31 No token in file: /data/token.txt
23/07/2023 16:09:31 Generating new token
23/07/2023 16:09:31 Failed connect to https://enlighten.enphaseenergy.com/login/login.json? to generate token part 1 got <Response [401]>  using this info {'user[email]': 'enter.enphase.userid@email.com', 'user[password]': 'enter.enphase.password'}
Traceback (most recent call last):
  File "//envoy_to_mqtt_json.py", line 577, in <module>
    main()
  File "//envoy_to_mqtt_json.py", line 568, in main
    if envoy_version == 7:
       ^^^^^^^^^^^^^
NameError: name 'envoy_version' is not defined
23/07/2023 16:09:31 Connected to 192.168.1.93:1883
23/07/2023 16:09:31 Subscribed to MQTT_TOPIC: /envoy/json
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
```

The add-on stops running after a few minutes. I'm quite sure last time that even after hours, the logs didn't populate and the add-on kept running. It looks to me as if firmware version R5.0.55 does not get recognised as a V5 firmware?

vk2him, 2023-07-23:
Thanks for that - yes your assumption is correct - it's not being recognised as v5  - Your's is R5 while mine is D5 - weird ... I'll make some changes - thanks

vk2him, 2023-07-23:
Ok, try 1.0.12 as that should now work for you

alexeiw123, 2023-07-23:
Amazing, that was very fast.

Updated and working on R5 firmware. My dashboard is humming along at 1s
intervals again.

vk2him, 2023-07-23:
No problem - I just made a minor change to display your actual firmware version - do you mind trying this changed version - since it's only a minor one it won't prompt you to reinstall so can you manually click the Check for Updates icon in addons, then clicking rebuild from within my addon. It will pull the new version and you should see this in your log:

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
23/07/2023 17:36:55 Serial number: <snip>
23/07/2023 17:36:55 Detected Firmware version R5
23/07/2023 17:36:55 FREEDS is inactive
23/07/2023 17:36:55 Read token from file /data/token.txt :  <Snip>
23/07/2023 17:36:55 Connected to 192.168.1.74:1883
23/07/2023 17:36:55 Subscribed to MQTT_TOPIC: envoy/json
23/07/2023 17:36:55 Envoy password is <snip>

alexeiw123, 2023-07-23:
Another update even though I'm already on 1.0.12?

This is @ snippit of what I have before clicking rebuild:

```
23/07/2023 17:34:06 Serial number: 
23/07/2023 17:34:06 Detected Firmware version 5
23/07/2023 17:34:06 FREEDS is inactive
23/07/2023 17:34:06 No token in file: /data/token.txt
23/07/2023 17:34:06 Generating new token
23/07/2023 17:34:06 Failed connect to https://enlighten.enphaseenergy.com/login/login.json? to generate token part 1 got <Response [401]>  using this info {'user[email]': 'enter.enphase.userid@email.com', 'user[password]': 'enter.enphase.password'}
23/07/2023 17:34:06 Connected to 192.168.1.93:1883
23/07/2023 17:34:06 Subscribed to MQTT_TOPIC: /envoy/json
23/07/2023 17:34:06 Envoy password is
```

And after 

```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
23/07/2023 18:08:07 Serial number: 
23/07/2023 18:08:07 Detected Firmware version R5
23/07/2023 18:08:07 FREEDS is inactive
23/07/2023 18:08:07 No token in file: /data/token.txt
23/07/2023 18:08:07 Generating new token
23/07/2023 18:08:07 Failed connect to https://enlighten.enphaseenergy.com/login/login.json? to generate token part 1 got <Response [401]>  using this info {'user[email]': 'enter.enphase.userid@email.com', 'user[password]': 'enter.enphase.password'}
23/07/2023 18:08:07 Envoy password is
```

I think you might've already pushed that change by the time I did the last update.

vk2him, 2023-07-23:
Perfect thanks - yes I was looking for this change: ```Detected Firmware version R5```

closed 2023-07-23 by alexeiw123