vk2him/Enphase-Envoy-mqtt-json #20: HA 2023.7.2 split configuration help

enkrypt3d, 2023-07-23:
On V7 on the envoy and as of a few months ago you can't put any MQTT sensors in the configuration.yaml so I'm having a tough time getting this working. Here's my MQTT.yaml:

sensor:
    - name: envoy mqtt consumption
      state_topic: "envoy/json"
      value_template: '{{ value_json[1]["activePower"] | round(0) | int(0)}}'
      unique_id: envoy_mqtt_consumption
      qos: 0
      unit_of_measurement: "W"
      state_class: measurement
      device_class: power
    - name: envoy mqtt voltage
      state_topic: "envoy/json"
      value_template: '{{ value_json[1]["voltage"] | round(0) | int(0)}}'
      unique_id: envoy_mqtt_voltage
      qos: 0
      unit_of_measurement: "V"
      state_class: measurement
      device_class: voltage
    - name: envoy mqtt current
      state_topic: "envoy/json"
      value_template: '{{ value_json[1]["current"] | round(2)}}'
      unique_id: envoy_mqtt_current
      qos: 0
      unit_of_measurement: "A"
      state_class: measurement
      device_class: current
    - name: envoy mqtt power factor
      state_topic: "envoy/json"
      value_template: '{{ value_json[1]["pwrFactor"] | round(2)}}'
      unique_id: envoy_mqtt_power_factor
      qos: 0
      unit_of_measurement: "%"
      state_class: measurement
      device_class: power_factor
    - name: "Gas Meter Reading"
      state_topic: "metermon/777####"
      qos: 0
      value_template: "{{ value_json.Consumption }}"
      unit_of_measurement: "ft³"
      device_class: "gas"

And my configuration.yaml:

# Configure a default setup of Home Assistant (frontend, api, etc)
default_config:

http:
  server_port: 8123
  use_x_forwarded_for: true
  trusted_proxies:
    - 172.30.33.0/24
  ip_ban_enabled: true
  login_attempts_threshold: 5

# Text to speech
tts:
  - platform: google_translate

group: !include groups.yaml
automation: !include automations.yaml
script: !include scripts.yaml
scene: !include scenes.yaml
template: !include templates.yaml
sensor: !include sensors.yaml
mqtt: !include mqtt.yaml
#binary_sensor: !include binary_sensors.yaml
#switch: !include switch.yaml

homeassistant:
  customize: !include customize.yaml
  auth_mfa_modules:
    - type: totp

logger:
  default: info

binary_sensor:
  - platform: hydrawise
switch:
  - platform: hydrawise

I imagine I need to put some of  your code in my templates.yaml file?

vk2him, 2023-07-24:
You can put mqtt sensors into configuration yaml, it just needs to be in the "new" format that starts with ```mqtt:```

Here's mine from configuration.yaml - if you put them in mqtt.yaml they will work provided you have a reference to it in configuration.yaml

```

mqtt:
  sensor:
    - name: "mqtt_production"
      state_topic: "envoy/json"
      qos: 0
      unit_of_measurement: "W"
      value_template: '{% if is_state("sun.sun", "below_horizon")%}0{%else%}{{ value_json["production"]["ph-a"]["p"]  | float }}{%endif%}'
      state_class: measurement
      device_class: power

    - name: "mqtt_consumption"
      state_topic: "envoy/json"
      value_template: '{{ value_json["total-consumption"]["ph-a"]["p"] }}'
      qos: 0
      unit_of_measurement: "W"
      state_class: measurement
      device_class: power

    - name: "mqtt_power_factor"
      state_topic: "envoy/json"
      qos: 0
      unit_of_measurement: "%"
      value_template: '{{ value_json["total-consumption"]["ph-a"]["pf"] }}'
      state_class: measurement
      device_class: power_factor

    - name: "mqtt_voltage"
      state_topic: "envoy/json"
      qos: 0
      unit_of_measurement: "V"
      value_template: '{{ value_json["total-consumption"]["ph-a"]["v"] }}'
      state_class: measurement
      device_class: voltage

    - state_topic: "envoy/json"
      name: "mqtt_net_voltage"
      value_template: '{{ value_json["net-consumption"]["ph-a"]["v"] }}'
      qos: 0
      unit_of_measurement: "V"
      state_class: measurement
      device_class: voltage

```

vk2him, 2023-07-24:
P.S. - I just noticed myReadme file doesn't show the "new" method for v5 - it still shows the old method. I'll update it with what I posted above as the new method is needed for all mqtt sensors

enkrypt3d, 2023-07-24:
What about the split template config? Thanks!

vk2him, 2023-07-24:
> What about the split template config? Thanks!

That's what I meant in my comment above - currenty your mqtt.yaml code isn't being processed because you may not be referencing it.

I can see you have this: 

 ``` customize: !include customize.yaml ```  

however you don't seem to have this  

``` customize: !include mqtt.yaml ``` ?

enkrypt3d, 2023-07-24:
group: !include groups.yaml
automation: !include automations.yaml
script: !include scripts.yaml
scene: !include scenes.yaml
template: !include templates.yaml
sensor: !include sensors.yaml
mqtt: !include mqtt.yaml

It's there.... but it's not using "customize". I have my gas meter running thru MQTT and metermon / RTL_TCP and it works fine. 

I guess your code needs to be updated to the latest HA syntax?

vk2him, 2023-07-24:
Hi - I just wanted to clarify some points as I may be misinterpreting what you're saying.

The addon simply sends a json string to the Mqtt broker - thats all it does. 

If you want to use that json string, you need to split it into the discrete components, and in Home assistant you do that with templates. So when you say "I guess your code needs to be updated to the latest HA syntax?" , my addon doesn't need to be changed, the templates in Home assiatnt that you write need to be done correctly, and that's up you you.

My documentation in the readme file has some sample code home assistant templates and as I mentioned above, I just noticed that in the v5 section the templates still refer to the "old" method that HA used. The V7 ones are correct though.

Before you do anything like trying to get power wheel card etc working, I recommend you first confirm your mqtt sensors are working by using the States tab in Developer tools 

[image]

You should probably also use mqttexplorer to see what the raw json string looks like (mine is v5 so is different to your v7)

[image]

enkrypt3d, 2023-07-24:
Yes the MQTT broker works fine for my other entities, it's just confusing
how you have your configurations layed out. You have some of the MQTT stuff
split but not all of it as well as the template....

My attempt at getting template.yaml updated but it doesn't like it:
- sensor:
    name: envoy mqtt consumption
    value_template: '{{ value_json[1]["activePower"] | round(0) | int(0)}}'
    unique_id: envoy_mqtt_consumption
    qos: 0
    unit_of_measurement: "W"
    state_class: measurement
    device_class: power
- sensor:
    name: envoy mqtt voltage
    value_template: '{{ value_json[1]["voltage"] | round(0) | int(0)}}'
    unique_id: envoy_mqtt_voltage
    qos: 0
    unit_of_measurement: "V"
    state_class: measurement
    device_class: voltage
- sensor:
    name: envoy mqtt current
    value_template: '{{ value_json[1]["current"] | round(2)}}'
    unique_id: envoy_mqtt_current
    qos: 0
    unit_of_measurement: "A"
    state_class: measurement
    device_class: current
- sensor:
    name: envoy mqtt power factor
    value_template: '{{ value_json[1]["pwrFactor"] | round(2)}}'
    unique_id: envoy_mqtt_power_factor
    qos: 0
    unit_of_measurement: "%"
    state_class: measurement
    device_class: power_factor

[image: image.png]

vk2him, 2023-07-24:
Your templates are missing this in every case:  ```state_topic: "envoy/json" ```
Refer to the readme v7 templates - you're trying to overcomplicate it by splitting into template.yaml and dropping code along the way. Why not try deleting the template.yaml code and putting this into configuration.yaml instead cause it works 100%

```
configuration.yaml configuration examples For FW 7
mqtt:
  sensor:
    - name: envoy mqtt consumption
      state_topic: "envoy/json"
      value_template: '{{ value_json[1]["activePower"] | round(0) | int(0)}}'
      unique_id: envoy_mqtt_consumption
      qos: 0
      unit_of_measurement: "W"
      state_class: measurement
      device_class: power
    - name: envoy mqtt voltage
      state_topic: "envoy/json"
      value_template: '{{ value_json[1]["voltage"] | round(0) | int(0)}}'
      unique_id: envoy_mqtt_voltage
      qos: 0
      unit_of_measurement: "V"
      state_class: measurement
      device_class: voltage
    - name: envoy mqtt current
      state_topic: "envoy/json"
      value_template: '{{ value_json[1]["current"] | round(2)}}'
      unique_id: envoy_mqtt_current
      qos: 0
      unit_of_measurement: "A"
      state_class: measurement
      device_class: current
    - name: envoy mqtt power factor
      state_topic: "envoy/json"
      value_template: '{{ value_json[1]["pwrFactor"] | round(2)}}'
      unique_id: envoy_mqtt_power_factor
      qos: 0
      unit_of_measurement: "%"
      state_class: measurement
      device_class: power_factor
```

enkrypt3d, 2023-07-24:
With having the state_topic, that it errors out as well. as I said you have
a mixture of the legacy formatting along with the new.... I'm trying to
translate it to the new format.... if it were only as easy as moving it
into one file then I'd do that but the entire structure has to change....

vk2him, 2023-07-24:
As far as I've seen, the two can be used together - you can put some templates in configuration.yaml and others in templates.yaml

I'm not sure why you don't want to test with the snippet from the post above into configuration.yaml as it works 100%

enkrypt3d, 2023-07-24:
Just tried it and it has created the sensors but they're not getting data.
Where do I specify the enlighten login? Maybe that's the issue..... also
for some reason my gas meter is no longer working so I'm debugging that now
too :(

vk2him, 2023-07-24:
First things first

1) fix your gas meter
2) Enlighten login details are in the addon Configuration - if you haven't set them then it wont work on V7
3) after entering them, restart the addon and look at the Log tab - you should see the token and a message saying it Subscribed to the MQTT_TOPIC: envoy/json
4) Use Developer Tools to look at the new mqtt sensors

enkrypt3d, 2023-07-24:
Realized that if there are any MQTT configs in the configuration.yaml, it
ignores everything in mqtt.yaml. So it'd be very helpful to just follow the
new HA standard and keep everything in its own config for templates.yaml /
mqtt.yaml etc. Any time I try to have them in both places, it only
acknowledges one of them.

vk2him, 2023-07-24:
Like I said earlier - writing the templates it your responsibility. I provided some template examples - use them or not - it's up to you.

enkrypt3d, 2023-07-24:
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
24/07/2023 01:00:01 Serial number: 202117004813
24/07/2023 01:00:01 Detected FW version 7
24/07/2023 01:00:01 FREEDS is inactive
24/07/2023 01:00:01 No token in file: /data/token.txt
24/07/2023 01:00:01 Generating new token
24/07/2023 01:00:01 Failed connect to https://enlighten.enphaseenergy.com/login/login.json? to generate token part 1 got <Response [401]>  using this info {'user[email]': 'xxxxxxxxxxx@email.com', 'user[password]': 'xxxxxxxxxxxx'}
24/07/2023 01:00:01 Generating new token
Exception in thread Thread-2 (scrape_stream_meters):
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "//envoy_to_mqtt_json.py", line 355, in scrape_stream_meters
    headers = {"Authorization": "Bearer " + ENVOY_TOKEN}
                                ~~~~~~~~~~^~~~~~~~~~~~~
TypeError: can only concatenate str (not "NoneType") to str
24/07/2023 01:00:01 Connected to 192.168.1.5:1883
24/07/2023 01:00:01 Subscribed to MQTT_TOPIC: /envoy/json
24/07/2023 01:00:01 Failed connect to https://enlighten.enphaseenergy.com/login/login.json? to generate token part 1 got <Response [401]>  using this info {'user[email]': xxxxxxxxx@email.com', 'user[password]': 'xxxxxxxxxxxxx'}
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

######## 

your addon wont start even with the correct values....

enkrypt3d, 2023-07-24:
24/07/2023 01:03:15 No token in file: /data/token.txt
24/07/2023 01:03:15 Generating new token
24/07/2023 01:03:15 Failed connect to https://enlighten.enphaseenergy.com/login/login.json? to generate token part 1 got <Response [401]>  using this info {'user[email]': 'xxxxxxxx@email.com', 'user[password]': 'xxxxxxxxxxxxxxxx'}
24/07/2023 01:03:15 Generating new token
Exception in thread Thread-2 (scrape_stream_meters):
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "//envoy_to_mqtt_json.py", line 355, in scrape_stream_meters
    headers = {"Authorization": "Bearer " + ENVOY_TOKEN}
                                ~~~~~~~~~~^~~~~~~~~~~~~
TypeError: can only concatenate str (not "NoneType") to str
24/07/2023 01:03:15 Connected to 192.168.1.5:1883
24/07/2023 01:03:15 Subscribed to MQTT_TOPIC: /envoy/json
24/07/2023 01:03:15 Failed connect to https://enlighten.enphaseenergy.com/login/login.json? to generate token part 1 got <Response [401]>  using this info {'user[email]': 'xxxxxxxx@email.com', 'user[password]': 'xxxxxxxxxxxx'}
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

vk2him, 2023-07-24:
Please confirm you have actually entered your enlighten username and password in the ephase_user and enphase_user_pass configuration tab? The 401 error you posed looks suspiciously like you're using the default values and not yours?

[image]

enkrypt3d, 2023-07-24:
Yes all the data is entered correctly

vk2him, 2023-07-24:
I'll raise this in https://github.com/vk2him/Enphase-Envoy-mqtt-json/issues/5

helderfmf, 2023-07-24:
@enkrypt3d 
Can you login here https://enlighten.enphaseenergy.com/login with your credentials? 
If yes, can you see data from you Envoy there, like readings, history, etc? Then go to https://entrez.enphaseenergy.com and login again and try to generate a token, just start typing 1000 in the "Select System *" it yours will apear.

enkrypt3d, 2023-07-24:
Yes I'm using the same account with other enphase integrations and it's
working with other integrations ...

enkrypt3d, 2023-07-24:
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
24/07/2023 16:36:11 Serial number: 202117004813
24/07/2023 16:36:11 Detected FW version 7
24/07/2023 16:36:11 FREEDS is active, using topic: Inverter/GridWatts
24/07/2023 16:36:11 No token in file: /data/token.txt
24/07/2023 16:36:11 Generating new token
24/07/2023 16:36:11 Failed connect to
https://enlighten.enphaseenergy.com/login/login.json? to generate
token part 1 got <Response [401]>  using this info {'user[email]':

'xxxxxxActualEnphasePassword'}
24/07/2023 16:36:11 Generating new token
24/07/2023 16:36:11 Connected to 192.168.1.5:1883
24/07/2023 16:36:11 Subscribed to MQTT_TOPIC: /envoy/json
24/07/2023 16:36:11 Failed connect to

I thought the token generation URL was
https://entrez.enphaseenergy.com/login not
https://enlighten.enphaseenergy.com? I can login just fine to both....

vk2him, 2023-07-24:
@enkrypt3d 

This is strange - it appears you haven't typed your password in correctly into the addon's configuration, or that the sequence of password characters you're using is causing an issue with the python script.

The addon code generates the token in two separate section - the first one scrapes some data when you log on to   https://enlighten.enphaseenergy.com/login - this is the page that displays an overview of your system and is purely used to obtain the login details needed for the next section.

The next section takes the login details and plugs them into https://entrez.enphaseenergy.com/ which then generates your token.

From your error message, section 1 returns <Response [401]> which basically means https://enlighten.enphaseenergy.com/login  is saying your username/password is incorrect.

So for testing, can I please ask you do two things?
1) Triple check you have entered your password/user email correctly into the addon configuration. 
2) If it still fails with 401 after your confirm the password is correct, change your password on Enphase to be alpha numeric with no special letters like !@#$%^&*(){}[] etc ..... then put that new password into the addon configuration and report back here?

enkrypt3d, 2023-07-24:
Instead of changing the password I just escaped the special characters in
the password field. it'd be nice if this was added and also masked?

Now the add on is running. still trying to get the template & mqtt yaml
code sorted out. Entities still in "unknown" state. it's a bit confusing to
me....

enkrypt3d, 2023-07-24:
Well it starts and then has the same error again a few minutes later...
changing the password would break all my other enphase stuff I have
running.......

enkrypt3d, 2023-07-24:
Tried changing my password and it eventually fails with the same error:

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
24/07/2023 19:42:00 Serial number: 202117004813
24/07/2023 19:42:00 Detected FW version 7
24/07/2023 19:42:00 FREEDS is active, using topic: Inverter/GridWatts
24/07/2023 19:42:00 No token in file: /data/token.txt
24/07/2023 19:42:00 Generating new token
24/07/2023 19:42:00 Failed connect to
https://enlighten.enphaseenergy.com/login/login.json? to generate
token part 1 got <Response [401]>  using this info {'user[email]':
***@***.***', 'user[password]': 'xxxxxxxxxxx'}
24/07/2023 19:42:00 Generating new token
24/07/2023 19:42:00 Connected to 192.168.1.5:1883
24/07/2023 19:42:00 Subscribed to MQTT_TOPIC: /envoy/json
Exception in thread Thread-2 (scrape_stream_meters):
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "//envoy_to_mqtt_json.py", line 355, in scrape_stream_meters
    headers = {"Authorization": "Bearer " + ENVOY_TOKEN}
                                ~~~~~~~~~~^~~~~~~~~~~~~
TypeError: can only concatenate str (not "NoneType") to str
24/07/2023 19:42:00 Failed connect to
https://enlighten.enphaseenergy.com/login/login.json? to generate
token part 1 got <Response [401]>  using this info {'user[email]':
***@***.***', 'user[password]': 'xxxxxxxxxx'}
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

enkrypt3d, 2023-07-25:
Where can I copy my token to? I already have a 1 year token so can you add
a field for existing tokens?

vk2him, 2023-07-25:
The token is stored in the container that the addon creates, so you'll need Portainer or another method to access the container. The token is stored in /data/token.txt within the container (not /data/token in the HA OS)

enkrypt3d, 2023-07-25:
Seems this doesn't work either - what format does it need to be in? I
tried including the {"auth_token": "token"} and got the same
errors....

s6-rc: info: service legacy-services successfully started
24/07/2023 21:59:28 Serial number: 202117004813
24/07/2023 21:59:28 Detected FW version 7
24/07/2023 21:59:28 FREEDS is active, using topic: Inverter/GridWatts
24/07/2023 21:59:28 Read token from file /data/token.txt :
"eyJ******************************************************Olk75Sw"

Exception in thread Thread-2 (scrape_stream_meters):
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "//envoy_to_mqtt_json.py", line 356, in scrape_stream_meters
    stream = requests.get(url, timeout=5, verify=False, headers=headers)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/requests/api.py", line 73, in get
    return request("get", url, params=params, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/requests/api.py", line 59, in request
    return session.request(method=method, url=url, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/requests/sessions.py", line
589, in request
    resp = self.send(prep, **send_kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/requests/sessions.py", line
703, in send
    r = adapter.send(request, **kwargs)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/requests/adapters.py", line
486, in send
    resp = conn.urlopen(
           ^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/urllib3/connectionpool.py",
line 703, in urlopen
    httplib_response = self._make_request(
                       ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/urllib3/connectionpool.py",
line 398, in _make_request
    conn.request(method, url, **httplib_request_kw)
  File "/usr/lib/python3.11/site-packages/urllib3/connection.py", line
244, in request
    super(HTTPConnection, self).request(method, url, body=body, headers=headers)
  File "/usr/lib/python3.11/http/client.py", line 1286, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/usr/lib/python3.11/http/client.py", line 1327, in _send_request
    self.putheader(hdr, value)
  File "/usr/lib/python3.11/site-packages/urllib3/connection.py", line
224, in putheader
    _HTTPConnection.putheader(self, header, *values)
  File "/usr/lib/python3.11/http/client.py", line 1264, in putheader
    raise ValueError('Invalid header value %r' % (values[i],))
ValueError: Invalid header value b'Bearer
"eyJr*******************************************************************************************kROlk75Sw"\n'
24/07/2023 21:59:28 Connected to 192.168.1.5:1883
24/07/2023 21:59:28 Subscribed to MQTT_TOPIC: /envoy/json
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

vk2him, 2023-07-25:
It's the full token on a single line without a newline character at the end  - so using your error and assuming your token starts with eyJr and ends with kROlk75Sw it should be a single line with exactly this:

```eyJr*******************************************************************************************kROlk75Sw```

vk2him, 2023-07-25:
Can you also please advise the version number of my addon you're running, just in case you have an old variant somehow? Something weird is happening with your system ...

enkrypt3d, 2023-07-25:
Using  Current version: 1.0.12 - so no quotations just the token by itself
correct?

vk2him, 2023-07-25:
Correct - 1.0.12 is the current version as well - maybe try rebuilding it as a clutching straws effort

enkrypt3d, 2023-07-25:
What does USE_FREEDS do?

Also note I'm on IQ Gateway Software Version 7.3.517 if that helps

vk2him, 2023-07-25:
I wrote the V5 part of the code as I have a v5 Envoy. Initially I didn't want to support V7 as I had no way of testing. Helder @helderfmf managed to get v7 working using my v5 code as a base. I merged his v7 code into mine last week.

He added some code specific for his use case which included Freeds - it's a low cost method to use excess solar generation to heat water or a floor slab. Just leave the Freed switch off if you don't need it. 

You'll see I refer some questions to Helder if they are specific to v7 code that I can't test.

vk2him, 2023-07-25:
What I can't understand is why you were able to get some data out of this and now it stopped with the error you showed above ... i can't test as I don't have v7 so hopefully Helder may give some insight

vk2him, 2023-07-27:
@enkrypt3d  - Helder has added some debug code that will help working out why it doesn't work for you. As this is testing, I don't want to merge the debug code into the Public released version. Can I ask you to assist please to test Helders code by adding his repo to HA, adding his test version and then report back here with the log from the addon? https://github.com/helderfmf/Enphase-Envoy-mqtt-json
Thanks

vk2him, 2023-07-28:
@enkrypt3d  - Hi, can you please try updating to my version (rather than Helders) as some debug code has been added today to hopefully identify where the issues may be. Turn on the Debug switch in Configuration and copy/paste the log here please after you remove your long token. Then turn off debug as it can fill the log quickly. Thanks

enkrypt3d, 2023-07-28:
Ok I can take a look and let you know thanks!

vk2him, 2023-07-28:
@enkrypt3d  - the error you're seeing was also reported in this issue https://github.com/vk2him/Enphase-Envoy-mqtt-json/issues/18 - it was resolved in the last few posts in the thread.

Take a read as the fix may help you - you may have a line feed character at the end of your token in your token.txt file.

closed 2023-08-18 by vk2him