vk2him/Enphase-Envoy-mqtt-json #22: unable to start

foXaCe, 2023-07-27:
hello, I use the home assistant addon and I can't go further than that
I already use mqtt with other addons without problem

my config

MQTT_HOST: 192.168.1.5
MQTT_USER: ****
MQTT_PASSWORD: ***
MQTT_PORT: 1883
MQTT_TOPIC: /envoy/json
ENVOY_HOST: 192.168.1.11
ENVOY_USER: ****
ENVOY_USER_PASS: ****
USE_FREEDS: false

and the log

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
27/07/2023 18:56:35 Serial number: ******
27/07/2023 18:56:35 Detected FW version 7
27/07/2023 18:56:35 FREEDS is inactive
27/07/2023 18:56:35 No token in file: /data/token.txt
27/07/2023 18:56:35 Generating new token
27/07/2023 18:56:35 Token generated 
*********************

Thanks for your help

vk2him, 2023-07-28:
Hi, can you please try updating as some debug code has been added to hopefully identify where the issues may be. Turn on the Debug switch in Configuration and copy/paste the log here please after you remove your long token. Then turn off debug as it can fill the log quickly. Thanks

foXaCe, 2023-07-28:
hi, thanks for the debug option
first i understood the problem
it was necessary to put in the creation of the sensor

That
      state_topic: "/envoy/json"

instead of that

      state_topic: "envoy/json"

he miss for me the /  🥲

maybe in the future a custom component to create sensors more easily for noobs 😊

thank for your job 👌

vk2him, 2023-07-28:
Glad you got it sorted, yes it would be nice to have a component to create sensors .... one day

closed 2023-07-28 by vk2him