vk2him/Enphase-Envoy-mqtt-json #23: Error 503 after Envoy reboot

Turbox35, 2023-07-31:
Hello,

Yesterday I had an issue with one inverter and I need to reboot the Envoy.
The plugin stop working with an Error 503. It seems that the token was not valid after the reboot but the plugin consider it ok in the .txt file.
To solve the problem, I had to uninstall the plugin and re-install it.

Is it possible to add a validity token check or if an error occurs, try 15min later and/or regenerate a new token?

vk2him, 2023-07-31:
Hi, this is very strange, working out why your token was invalid will help the solution.  We also need to cater for the situation where power goes off and/or internet is not available when the envoy reboots so it has to use the stored token. 

The tokens that are generated and stored have a 12 month validity unless your Enphase ID is an "installer", in which case it's only valid for 12 hours.

Other than the time period, the things that would invalidate the token are a change in envoy serial number/your Enphase email address changes or the envoy firmware level changing.

Are you able to check if the Envoy firmware level was updated after you rebooted it? Do you know if your email is an installer or owner? If you're unsure, decoding the token will tell you.

Use this site to decode your token - https://jwt.io/  - paste your Token into the JWT string box and look at the Payload box. With credit to @del13r here is a screenshot of what to look for:

[image]

Can you please advise what the "enphaseUser" is - is it installer or is it owner ?
Can you also hover your cursor over the exp field per the screenshot - this will show the expiry date and it should be 12 months away.

Do you have a backup of the token.txt file before you reinstalled the addon as decoding the old one will also be helpful to resolve this. 
Thanks

Turbox35, 2023-07-31:
I am not installer (even if I have install the system myself).
[image]
I don't have a backup of the token.txt and I don't know if the envoy made a firmware update but it seems I am up-to-date.
Perhaps it was not a token issue. Do you already see this 503 error in the past?
If your Envoy reboot, the plugin restart to stream automatically?

vk2him, 2023-07-31:
I have a v5 system so I'm unable to test.
@helderfmf  - Helder, have you any insight into this please? I don't think the token was the problem unless the Envoy was coincidently updated during the reboot. Have you seen 503 errors after rebooting your Envoy? Thanks

helderfmf, 2023-07-31:
Hi there, 503 is "sever not available", if it was a problem with the token you should get 401, and in that case the addon should try to generate a new one.

I got some 503 when doing debug with multiple connections to Envoy, of running 2 scripts at the same time, one via addon and the other with python directly. So, unless you have 2 or more addons/scripts running, I don't see the reason to give 503.
One other possibility could be, and I cannot verify this as my system has only a meter on the Grid and doesn't control any inverter, Envoy is not responding in a timely manner because of too many components, and could generate this 503, but if this is the case, it should ****not** always** give 503, some valid responses should apear. 

One more thing, an installer login cannot read information from an "commissioned" envoy, unless is his own installation. What this means is that, for example mine was installed by a company, and they are the installers, they give me access as owner to my envoy, even if I try to connect with other account as a installer to my system it is denied. Thats why I cannot access /stream/meters with any account, my original one as owner, and a new one I created as installer with the 12h validity:
[image]

Turbox35, 2023-08-01:
Oh, it's probably my issue because, yes, I use multiple connections to my envoy (1 for domoticz plugin, 1 for domoticz beta plugin and 1 for Home Assistant plugin).
At the end (I hope) I will use only one connection which will feed all my domotic systems using MQTT Auto Discovery.

closed 2023-08-07 by Turbox35