vk2him/Enphase-Envoy-mqtt-json #25: Allow V8 firmware

OzGav, 2023-09-12:
Just got my system installed today with brand new hear. The offical integration is working but I am hoping to get info from my 3 phases in real time from this one. I am seeing this message:

`Cannot match firmware version, got  ['D8.2.22']`

Can we allow firmware >=7

vk2him, 2023-09-12:
Hi, I've merged your PR - let us know how it goes?

OzGav, 2023-09-12:
Will do. I'm watching for the addon update....

vk2him, 2023-09-12:
It won't show as an update as you only changed the python code - try deleting and reinstalling as it will then pull the updated version. By the way, are you Ozgav from Music assistant?

OzGav, 2023-09-12:
OK and yep :-)

OzGav, 2023-09-12:
Yes I remember you as well. It is good isn't it. When I sit down and just use it rather than test it I go "This is so good!" :-)

OzGav, 2023-09-14:
Addon working perfectly with the v8 firmware! Thanks I'm finding it more to my needs that the offical integration.

closed 2023-09-14 by OzGav