vk2him/Enphase-Envoy-mqtt-json #28: Sample Config MQTT Sensors for V7 and V8 firmware incorrect

goostincain, 2023-12-01:
Hey there. Thanks for your work. 

I have been trying to get the MQTT sensors to work in Home Assistant and it wasn't working. I had to change the following to make it work for me:

From
state_topic: "envoy/json"

To
state_topic: "/envoy/json"

the leading slash is needed.

vk2him, 2023-12-01:
Hey, thanks for using the addon and the message.

I can see what's happened here - someone reported a while ago that if the envoy mqtt topic starts with a leading slash, if you use mqtt explorer then the envoy topic is indented and doesn't look "logical". By removing the leading slash, it appears in the far left column as seen in this example:

[image]

If you include the slash it looks like this which is indented into a "blank" topic:

[image]

I modified the sensor templates in the readme file to remove the slash from the topic, but I didn't remove the slash from the default configuration within the add-on which caused this issue for you. Instead of modifying the sensors by adding the slash, you can remove the slash from the addon configuration : 

[image]

I'll modify the readme shortly so the sensors have the slash which matches the addon configuration, as if I change the default configuration by removing the slash, it will be a breaking change for anyone who like you have sensors that require the slash.

Sorry for the hassle, cheers

goostincain, 2023-12-01:
Sounds good, and makes sense. There’s nothing quite like banging your head on a keyboard for a slash! ;)Justin

image.png (view on web)
image.png (view on web)
image.png (view on web)

leah-potato, 2024-05-10:
That one got me too! Should have checked here first 🤦‍♀️

networkerict, 2024-08-21:
Argggh...... I have been troubleshooting it for 2 hours before deciding to check here. It was driving me crazy. It works now as expected.... ;-)

sigh....

vk2him, 2024-08-22:
Sorry for the very long delay in me updating the documentation, and the frustration caused to you to fault find. I just updated the documentation so the template snow have

```state_topic: "/envoy/json"```

closed 2024-08-22 by vk2him