vk2him/Enphase-Envoy-mqtt-json #30: Enphase upgrade sensors/reporting stopped working

timolow, 2024-01-15:
Howdy,

Seems that my enphase got upgraded and I lost all monitoring data, I got help here setting up the sensors correctly https://github.com/vk2him/Enphase-Envoy-mqtt-json/issues/7

Here is my current config:
`sensor 1001:
  - platform: integration
    source: sensor.solarpower
    name: energy_solar
    method: left
##    unit: kWh
    unit_prefix: k
    unit_time: h
    round: 2
sensor 1002:
  - platform: integration
    source: sensor.exporting
    name: energy_exporting
    method: left
#    unit: kWh
    unit_prefix: k
    unit_time: h
    round: 2
sensor 1003:
  - platform: integration
    source: sensor.importing
    name: energy_importing
    method: left
#    unit: kWh
    unit_prefix: k
    unit_time: h
    round: 2`

`mqtt:
  sensor:
    - name: envoy mqtt consumption
      state_topic: "/envoy/json"
      value_template: '{{ value_json[1]["activePower"] | round(0) | int(0)}}'
      unique_id: mqtt_consumption
      qos: 0
      unit_of_measurement: "W"
      state_class: measurement
      device_class: power
    - name: envoy mqtt voltage
      state_topic: "/envoy/json"
      value_template: '{{ value_json[1]["voltage"] | round(0) | int(0)}}'
      unique_id: mqtt_voltage
      qos: 0
      unit_of_measurement: "V"
      state_class: measurement
      device_class: voltage
    - name: envoy mqtt current
      state_topic: "/envoy/json"
      value_template: '{{ value_json[1]["current"] | round(2)}}'
      unique_id: mqtt_current
      qos: 0
      unit_of_measurement: "A"
      state_class: measurement
      device_class: current
    - name: envoy mqtt power factor
      state_topic: "/envoy/json"
      value_template: '{{ value_json[1]["pwrFactor"] | round(2)}}'
      unique_id: mqtt_power_factor
      qos: 0
      unit_of_measurement: "%"
      state_class: measurement
      device_class: power_factor
      
sensor:
  
  #
  # These ones are for Envoy via mqtt
  #

  - platform: template
    sensors:
      exporting:
        friendly_name: "Current MQTT Energy Exporting"
        value_template: "{{ [0, (states('sensor.mqtt_production') | int(0) - states('sensor.mqtt_consumption') | int(0))] | max }}"
        unit_of_measurement: "W"
        icon_template: mdi:flash
      importing:
        friendly_name: "Current MQTT Energy Importing"
        value_template: "{{ [0, (states('sensor.mqtt_consumption') | int(0) - states('sensor.mqtt_production') | int(0))] | max }}"
        unit_of_measurement: "W"
        icon_template: mdi:flash
      solarpower:
        friendly_name: "Solar MQTT Power"
        value_template: "{{ states('sensor.mqtt_production')}}"
        unit_of_measurement: "W"
        icon_template: mdi:flash
`

vk2him, 2024-01-16:
Hi Tim,
Can you please share the output from the log here.

Also, please advise which Software Version you now have by looking at the bottom of here http://envoy.local/ under the "Envoy" heading

timolow, 2024-01-16:
Howdy!

The Envoy version that got upgraded to is D7.3.130 (91f206)

The log for the plugin is as follows:
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
16/01/2024 09:10:34 Serial number: 121804016771
16/01/2024 09:10:34 Detected FW version 7
16/01/2024 09:10:34 FREEDS is inactive
16/01/2024 09:10:34 Read token from file /data/token.txt :  <SNIP>
```

vk2him, 2024-01-16:
Thanks, that looks correct, the script is talking to your Envoy and is getting the correct responses from it. 

Can you please check your configuration from the plugin - you should have the following under Mqtt topic to match the Sensor you created;

```MQTT_TOPIC: /envoy/json```

timolow, 2024-01-17:
Here is the mqtt topic: /envoy/json

`[{"eid": 704643328, "timestamp": 1705453271, "actEnergyDlvd": 55306378.348, "actEnergyRcvd": 0.029, "apparentEnergy": 71144746.814, "reactEnergyLagg": 24082724.25, "reactEnergyLead": 0.011, "instantaneousDemand": -0.0, "activePower": -0.0, "apparentPower": 501.18, "reactivePower": 496.191, "pwrFactor": 0.0, "voltage": 241.378, "current": 4.152, "freq": 60.0, "channels": [{"eid": 1778385169, "timestamp": 1705453271, "actEnergyDlvd": 27722305.637, "actEnergyRcvd": 0.028, "apparentEnergy": 35682339.543, "reactEnergyLagg": 12161748.136, "reactEnergyLead": 0.005, "instantaneousDemand": -0.0, "activePower": -0.0, "apparentPower": 250.549, "reactivePower": 247.859, "pwrFactor": 0.0, "voltage": 120.74, "current": 2.075, "freq": 60.0}, {"eid": 1778385170, "timestamp": 1705453271, "actEnergyDlvd": 27584072.71, "actEnergyRcvd": 0.001, "apparentEnergy": 35462407.271, "reactEnergyLagg": 11920976.114, "reactEnergyLead": 0.005, "instantaneousDemand": 0.0, "activePower": 0.0, "apparentPower": 250.631, "reactivePower": 248.332, "pwrFactor": 0.0, "voltage": 120.638, "current": 2.078, "freq": 60.0}, {"eid": 1778385171, "timestamp": 1705453271, "actEnergyDlvd": 0.0, "actEnergyRcvd": 0.0, "apparentEnergy": 0.0, "reactEnergyLagg": 0.0, "reactEnergyLead": 0.0, "instantaneousDemand": 0.0, "activePower": 0.0, "apparentPower": 0.0, "reactivePower": 0.0, "pwrFactor": 0.0, "voltage": 0.0, "current": 0.0, "freq": 60.0}]}, {"eid": 704643584, "timestamp": 1705453271, "actEnergyDlvd": 105905125.185, "actEnergyRcvd": 18017277.518, "apparentEnergy": 140367853.981, "reactEnergyLagg": 4065718.823, "reactEnergyLead": 28181399.512, "instantaneousDemand": 4839.408, "activePower": 4839.408, "apparentPower": 5050.778, "reactivePower": -316.181, "pwrFactor": 0.951, "voltage": 241.496, "current": 41.85, "freq": 60.0, "channels": [{"eid": 1778385425, "timestamp": 1705453271, "actEnergyDlvd": 20136900.094, "actEnergyRcvd": 14878076.158, "apparentEnergy": 42030665.776, "reactEnergyLagg": 3427997.134, "reactEnergyLead": 8919744.426, "instantaneousDemand": 2023.834, "activePower": 2023.834, "apparentPower": 2055.482, "reactivePower": 303.062, "pwrFactor": 0.984, "voltage": 120.775, "current": 16.983, "freq": 60.0}, {"eid": 1778385426, "timestamp": 1705453271, "actEnergyDlvd": 85768225.091, "actEnergyRcvd": 3139201.36, "apparentEnergy": 98337188.205, "reactEnergyLagg": 637721.689, "reactEnergyLead": 19261655.086, "instantaneousDemand": 2815.574, "activePower": 2815.574, "apparentPower": 2995.296, "reactivePower": -619.244, "pwrFactor": 0.928, "voltage": 120.721, "current": 24.867, "freq": 60.0}, {"eid": 1778385427, "timestamp": 1705453271, "actEnergyDlvd": 0.0, "actEnergyRcvd": 0.0, "apparentEnergy": 0.0, "reactEnergyLagg": 0.0, "reactEnergyLead": 0.0, "instantaneousDemand": 0.0, "activePower": 0.0, "apparentPower": 0.0, "reactivePower": 0.0, "pwrFactor": 0.0, "voltage": 0.0, "current": 0.0, "freq": 60.0}]}]`

vk2him, 2024-01-17:
Can you please confirm your configuration within the addon is set to /envoy/json and not envoy/json ?

timolow, 2024-01-17:
Here is the yaml portion of the config you asked for:
MQTT_TOPIC: /envoy/json

vk2him, 2024-01-17:
The mqtt topic you posted above shows that the addon is correctly reading the Envoy and is republishing it as mqtt - this isn't an issue with the addon, it's something to do with your template and/or the sensors. I assume you've restarted HA after the Envoy was upgraded to version 7? You might need to do some fault-finding using the Developer Tools in HA under STATES for your sensors:
[image]

vk2him, 2024-02-16:
Hi Tim - are you still having issues?

vk2him, 2024-02-27:
This should work, thanks to @alexeiw123

https://github.com/vk2him/Enphase-Envoy-mqtt-json/issues/29#issuecomment-1965717646

closed 2024-02-27 by vk2him

timolow, 2024-02-28:
#29 fixed the issue for me

alexeiw123, 2024-02-28:
Great to hear.

I realised eventually that `value_json[0]` refers to the production CT values and `value_json[1]` probably refers to the consumption CT values. As my consumption CT (like most) is the entire load including the solar, it returns the premise net values. Once you know what the two values are, working out the load without generation is trivial.

It would be interesting to see if it changes based on the related configuration setting on the enphase installer dashboard.

alexeiw123, 2024-04-30:
Just thought I would drop an update here with obligatory link to #29 to keep it all referenced.

I found this fantastic tool: https://jsonpathfinder.com/ that spits out the json path you need to access each value.

I wanted to see individual phase consumption of my 3 phase system as well as the total house consumption, so added these entities:

```
  - name: "mqtt_consumption_ph1"
    state_topic: "/envoy/json"
    unit_of_measurement: "W"
    qos: 0
    state_class: measurement
    device_class: power
    value_template: "{{ ( value_json[1].channels[0].activePower + value_json[0].channels[0].activePower ) | round(0) | int(0) }}" # Net ph1 + production ph1

  - name: "mqtt_consumption_ph2"
    state_topic: "/envoy/json"
    unit_of_measurement: "W"
    qos: 0
    state_class: measurement
    device_class: power
    value_template: "{{ ( value_json[1].channels[1].activePower + value_json[0].channels[1].activePower ) | round(0) | int(0) }}" # Net ph2 + production ph2

  - name: "mqtt_consumption_ph3"
    state_topic: "/envoy/json"
    unit_of_measurement: "W"
    qos: 0
    state_class: measurement
    device_class: power
    value_template: "{{ ( value_json[1].channels[2].activePower + value_json[0].channels[2].activePower ) | round(0) | int(0) }}" # Net ph3 + production ph3
```

Thought they could be useful for someone out there

vk2him, 2024-04-30:
Nice find Alex, and thanks for the sample entities!