vk2him/Enphase-Envoy-mqtt-json #32: Max retries exceeded

leah-potato, 2024-05-12:
Hello! 

Thanks so much for setting all this up. When it's working it's amazing to have live readings, but running into an issue wither intermittent connectivity: 

`Exception fetching stream data: HTTPSConnectionPool(host='envoy.local', port=443): Max retries exceeded with url: /ivp/meters/readings (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0xffff99cb78d0>: Failed to establish a new connection: [Errno -5] Name has no usable address'))`

`s6-rc: info: service legacy-services successfully started
Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/urllib3/connection.py", line 174, in _new_conn
    conn = connection.create_connection(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/urllib3/util/connection.py", line 72, in create_connection
    for res in socket.getaddrinfo(host, port, family, socket.SOCK_STREAM):
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/socket.py", line 962, in getaddrinfo
    for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
socket.gaierror: [Errno -5] Name has no usable address`

**Enphase-Envoy-mqtt-json config:**
[image]
What does USE_FREEDS do? Currently turned off.

**Mosquitto broker config:** 
[image]

When it does work, I can see the values correctly in MQTT explorer. My enphase gateway is connected via wifi, but the connection is pretty rock solid with https://envoy.local/home always available. 

Appreciate any tips on how to troubleshoot. Thanks!

vk2him, 2024-05-12:
Hi Leah, 

I suspect it's a local DNS issue, try changing envoy.local to the actual envoys ip in the configuration then restart it. Coincidentally another user had a similar issue 2 weeks ago and reported it under issue 31.

leah-potato, 2024-05-12:
Yep, that was it! I wonder how I can get local dns working correctly. I suspect that is the issue for some devices on my network with moving ip addresses. 

Anyway thanks a bunch. So cool having real time data! I've turned into my dad going around turning off light switches and watching the gauge go down instantly. You've created a monster.

vk2him, 2024-05-12:
Glad you got it working, local dns can be a bit of a pita. 

I'm using a card I found to display the data, it's the Power Distribution card  [image]

'''

type: custom:power-distribution-card
title: Power
entities:
  - decimals: '0'
    display_abs: false
    name: Importing
    unit_of_display: W
    icon: mdi:transmission-tower
    entity: sensor.importing_mqtt
    preset: grid
    double_tap_action:
      undefined: {}
    icon_color:
      bigger: red
      smaller: red
      equal: green
    arrow_color:
      bigger: red
      smaller: red
      equal: green
  - decimals: 2
    display_abs: true
    name: home
    unit_of_display: W
    consumer: true
    icon: mdi:home-assistant
    entity: sensor.house
    preset: home
    double_tap_action: {}
    icon_color:
      bigger: red
      smaller: red
    arrow_color:
      bigger: red
      smaller: red
    invert_value: true
  - display_abs: true
    name: Solar
    unit_of_display: W
    icon: mdi:solar-power
    producer: true
    entity: sensor.solarpower_mqtt
    preset: solar
    double_tap_action:
      undefined:
        undefined: {}
    icon_color:
      bigger: orange
      equal: orange
      smaller: ''
    arrow_color:
      bigger: orange
      equal: orange
    decimals: '0'
  - decimals: '0'
    display_abs: true
    name: Exporting
    unit_of_display: W
    icon: mdi:transmission-tower
    preset: grid
    double_tap_action:
      undefined:
        undefined:
          undefined:
            undefined:
              undefined: {}
    entity: sensor.exporting_mqtt
    icon_color:
      bigger: red
      equal: green
      smaller: green
    color_threshold: '0'
    arrow_color:
      bigger: red
      equal: green
      smaller: green
    invert_value: true
    secondary_info_replace_name: false
  - decimals: 2
    display_abs: true
    name: placeholder
    unit_of_display: W
    preset: placeholder
  - decimals: '0'
    display_abs: true
    name: Hot Water
    unit_of_display: W
    consumer: true
    icon: mdi:water-boiler
    entity: sensor.fluksohwsconsumption
    preset: consumer
    double_tap_action:
      undefined: {}
    calc_excluded: true
    invert_value: true
    icon_color:
      bigger: ''
      smaller: teal
      equal: green
    arrow_color:
      bigger: blue
      smaller: teal
      equal: green
center:
  type: bars
  content:
    - preset: autarky
      name: autarky
      bar_color: Green
    - preset: ratio
      name: Ratio
      bar_color: Green
animation: flash
'''

leah-potato, 2024-05-15:
Pretty cool! I've been playing around with that power flow plus card but it doesn't support combining 3 phases. I'm really only just getting started with the dashboard and the first piece was getting real time data (thanks to you!) - means the we're not wasting a whole bunch of money exporting to the grid. Today was the first day where I've finally got it dialed in that we are only over/under ~100w. 

HA is very addictive.

[image]

vk2him, 2024-05-15:
What you've done is impressive. You should be able to create a sensor that sums each phase's power and use that in the Power flow plus card?

closed 2024-05-17 by vk2him

joergbattermann, 2025-02-08:
> Pretty cool! I've been playing around with that power flow plus card but it doesn't support combining 3 phases. I'm really only just getting started with the dashboard and the first piece was getting real time data (thanks to you!) - means the we're not wasting a whole bunch of money exporting to the grid. Today was the first day where I've finally got it dialed in that we are only over/under ~100w... at least during the day. I can see a battery in our future soon.
> 
> HA is very addictive I've come to realise.
> 
> <img alt="image" width="1429" src="https://private-user-images.githubusercontent.com/128001814/330774772-a84e170d-cfbc-4d9d-ab71-86a3f1f0209f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzkwNDgzMDYsIm5iZiI6MTczOTA0ODAwNiwicGF0aCI6Ii8xMjgwMDE4MTQvMzMwNzc0NzcyLWE4NGUxNzBkLWNmYmMtNGQ5ZC1hYjcxLTg2YTNmMWYwMjA5Zi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMjA4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDIwOFQyMDUzMjZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iMmIwZDc1NzcwNDBhNGFhNzI2MGFjOWEyMjIzMzBmZWJiYTA1OWUwMWEzZjc3YzdjZDEyNmQyMWUzYzAyNDhjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.hTJi3pFllsrhMZZtHlqez04xYDKKcwIB-PZS1UZjKaA">

@leah-potato .. do I see correctly that you have some sort of Enphase < > Tesla charging on excess solar set up? Is that something (very) custom by you or are you using some add-on/integration for that (& still working these days with Tesla's API changes etc)?