vk2him/Enphase-Envoy-mqtt-json #34: Cannot match firmware version

ellmout, 2024-08-23:
Hello,

Since this morning, it's not starting anymore :

```
23/08/2024 09:12:07 Cannot match firmware version, got  ['D8.2.4264']
Traceback (most recent call last):
23/08/2024 09:12:07 FREEDS is inactive
  File "//envoy_to_mqtt_json.py", line 143, in <module>
    if envoy_version != 5:
       ^^^^^^^^^^^^^
NameError: name 'envoy_version' is not defined
s6-rc: info: service legacy-services: stopping
```

Thx

ellmout, 2024-08-23:
Nevermind...

I just rebuild the image, and it's working again !

Sorry

closed 2024-08-23 by ellmout

jimmyeao, 2024-09-08:
I,m having this issue running the homeassistan add on, how did you rebuild?

dgaust, 2024-09-08:
In the add on page there is an option to rebuild

[image]