vk2him/Enphase-Envoy-mqtt-json #35: Inverter information stream

Turbox35, 2024-09-04:
I use a fork of this excellent add on for enphase system. Since few days, my firmware has changed to V8. 
My question is: do you know if now there is some additional stream/JSON related to micro inverter status? Because some information are missing like individual micro inverter power/production and I still use integrated Enphase add-on in parallel.
Or perhaps I miss something?

vk2him, 2024-09-05:
I have V5 so I'm unable to answer - you might take a look at this old issue in case it helps? https://github.com/vk2him/Enphase-Envoy-mqtt-json/issues/29#issuecomment-1844038706

Turbox35, 2024-09-07:
Thanks, but I can't find some such information.
Have a nice day.

closed 2024-09-07 by Turbox35