vk2him/Enphase-Envoy-mqtt-json #36: Add battery readings to MQTT stream

Underlyingglitch, 2024-09-18:
I set up this add-on because I want the live data from Enphase in Home Assistant. For the production/meter readings this is working just fine, but I don't see an option for battery storage. From the code it seems as if the data from /ivp/livedata/status is used, but in the MQTT packets no readings for storage are available. Also the structure of the JSON payload in the MQTT packets is not the same as the data on /ivp/livedata/status. 

Since the official Enphase integration in Home Assistant offers no live readings for batteries I was hoping maybe this add-on could solve that. I'd be happy to help develop/test it.

 /ivp/livedata/status has the following properties:
 
"meters": {
"pv":{}
"storage":{},
"grid":{},
"load":{},
"generator":{}
}

All with the following data:
"agg_p_mw": 339626,
"agg_s_mva": 534524,
"agg_p_ph_a_mw": 83880,
"agg_p_ph_b_mw": 128396,
"agg_p_ph_c_mw": 127350,
"agg_s_ph_a_mva": 117432,
"agg_s_ph_b_mva": 223298,
"agg_s_ph_c_mva": 193793

vk2him, 2024-09-18:
I'm not able to help sorry - I have V5 firmware with no batteries - all I can suggest is to use mqtt explorer https://mqtt-explorer.com/ and point it to your homeassistant broker. You can then see the json as created by this addon and from that, create a template to extract the required data using the examples I provided as a guide. This is what I see on version 5

[image]

Underlyingglitch, 2024-09-18:
I have changed the code slightly and used the scrape_stream_livedata() function for firmware version 8 now. It had a bug where the is_json_valid() function would fail because the data stream was encoded as a byte string. By updating the is_json_valid() function I solved this issue and it works.

Because the JSON result is different I added that bit to the README.md.

My code editor was complaining about the Callback API version 1 being deprecated (in the paho-mqtt). I ignored this for now, but this should probably be looked into. They now use function decorators but I couldn't get this fixed in the short time I tried.

Hope this helps!

closed 2024-09-19 by vk2him

vk2him, 2024-09-19:
Thanks for the work on this - I've merged it into the main. I might need to look into the deprecated Callback, although I've super busy with my day job so may not have a chance for a long time - if you have any luck I'd welcome any suggestions.
Cheers
Ian

PR #37 (2024-09-19): Added battery sensors for FW 8
Used the /ivp/livedata/stream endpoint to load live data for consumption, production, load and batteries. 

Closes #36