vk2him/Enphase-Envoy-mqtt-json #38: Not publicing to mqtt

npllange, 2024-10-15:
Did a home assistant setup. Al settings are set via the readme, but there is no publication to the mqqt broker. I checked the password, host adres and port several times. 

[image]

From the log i get : 

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
15/10/2024 09:05:52 Serial number: 1223081XXXXX
15/10/2024 09:05:52 Detected Firmware version D8
15/10/2024 09:05:52 FREEDS is inactive
15/10/2024 09:05:52 Read token from file data/token.txt :  eyJraWQiOiI3ZDEwMDA1ZC03ODk5LTRkMGQtYmNiNC0yNDRmOThlZTE1NmIiLCJ0eXAiOiJKV1QiLCJhbGciOiJFUzI1NiJ9.eyJhdWQiOiIxMjIzMDgxMTA4ODIiLCJpc3MiOiJFbnRyZXoiLCJlbnBoYXNlVXNlciI6Im93bmVyIiwiZXhwIjoxNzYwMzgyMDAwLCJpYXQiOjE3Mjg4NDYwMDAsIXXXXXXXXX
15/10/2024 09:05:52 Will wait for mqtt connect
15/10/2024 09:05:52 Finished waiting for mqtt connect
15/10/2024 09:05:52 Url: https://192.168.2.105/ivp/meters/readings
15/10/2024 09:05:52 headers: {'Authorization': 'Bearer eyJraWQiOiI3ZDEwMDA1ZC03ODk5LTRkMGQtYmNiNC0yNDRmOThlZTE1NmIiLCJ0eXAiOiJKV1QiLCJhbGciOiJFUzI1NiJ9.eyJhdWQiOiIxMjIzMDgxMTA4ODIiLCJpc3MiOiJFbnRyZXoiLCJlbnBoYXNlVXNlciI6Im93bmVyIiwiZXhwIjoxNzYwMzgyMDAwLCJpYXQiOjE3Mjg4NDYwMDAsIXXXXXXXXXXXX
15/10/2024 09:05:52  Connection refused - not authorised
Disconnect returned:
client: <paho.mqtt.client.Client object at 0x7f49cf017ce0>
userdata: None
result: 5
15/10/2024 09:05:52 stream: b'[\n    {\n        "eid": 704643328,\n        "timestamp": 1728976003,\n        "actEnergyDlvd": 523511.788,\n        "actEnergyRcvd": 3892.340,\n        "apparentEnergy": 623455.063,\n        "reactEnergyLagg": 92199.432,\n        "reactEnergyLead": 1263.230,\n        "instantaneousDemand": 671.630,\n        "activePower": 671.630,\n        "apparentPower": 713.117,\n        "reactivePower": 44.954,\n        "pwrFactor": 0.946,\n        "voltage": 688.024,\n        "current": 3.109,\n        "freq": 50.000,\n        "channels": [\n            {\n                "eid": 1778385169,\n                "timestamp": 1728976003,\n                "actEnergyDlvd": 196410.385,\n

npllange, 2024-10-16:
I found the problem. I made a (shameful) mistake by coping the mqqt password... It had a blank space on the beginning.  Thank you for your help and for making this add-on!

closed 2024-10-16 by npllange

vk2him, 2024-10-16:
No worries, I'm glad you found the problem- I think we've all made a similar mistake, just frustrating that it can take a while to notice it. Cheers