vk2him/Enphase-Envoy-mqtt-json #40: mqtt output with installation method 2

0berkampf, 2024-11-03:
I just have a question regarding the second installation method: when I proceed with the first installation method (as a HA add on), the resulting mqtt topic is like the one given in your example:

    {
        "eid": XXXXXXXXXX,
        "timestamp": 1689409016,
        "actEnergyDlvd": 0.063,
        "actEnergyRcvd": 7939.998,
        "apparentEnergy": 63680.783,
        "reactEnergyLagg": 788.493,
        …………..

But when I proceed with the second method, the resulting mqtt topic looks like this:

{
  "connection": {
    "mqtt_state": "connected",
    "prov_state": "configured",
    "auth_state": "ok",
    "sc_stream": "enabled",
    "sc_debug": "disabled"
  },
  "meters": {
    "last_update": 1730655384,
    "soc": 0,
    "main_relay_state": 1,
    "gen_relay_state": 5,
    "backup_bat_mode": 1,
    "backup_soc": 0,
    "is_split_phase": 0,
    "phase_count": 1,
    "enc_agg_soc": 0,
    "enc_agg_energy": 0,
    "acb_agg_soc": 0,
    "acb_agg_energy": 0,
    "pv": {
      "agg_p_mw": -5414,
      "agg_s_mva": -270677,
      "agg_p_ph_a_mw": -5414,
      ……………

Same enphase envoy system, same parameters for the configuration of your script but two different outputs published.
Did I miss something during the install or the configuration of the script?

Thx for your help.

0berkampf, 2024-11-03:
I think I found how to fix the issue, in the options.json file, I changed the line:
 "BATTERY_INSTALLED": **"False"**
by 
 "BATTERY_INSTALLED": **false**

vk2him, 2024-11-03:
Interesting - on my Linux Mint system I needed to use ```"BATTERY_INSTALLED": "False"```

On your system, did you also have to change the other options such as this ```  "ENVOY_HOST": "envoy.local", ```

0berkampf, 2024-11-03:
I’m using a RPI with Raspberry Pi OS, and no, only boolean options have to be changed…

Well, it’s working now as expected, thx for your answer.

closed 2024-11-03 by 0berkampf