vk2him/Enphase-Envoy-mqtt-json #41: Error message...possibly something with my mqtt broker settings

jgassens, 2024-12-15:
I have been scratching my head over this… My logs show the following:

> 15/12/2024 10:45:15 Serial number: 2**********1
> 15/12/2024 10:45:15 Detected Firmware version D8
> 15/12/2024 10:45:15 FREEDS is inactive
> 15/12/2024 10:45:15 Read token from file data/token.txt :  eyJraWQiOiI3ZDEwMDA1ZC03ODk5LTRkMGQtYmNiNC0yNDRmOThlZTE1NmIiLCJ0eXAiOiJKV1QiLCJhbGciOiJFUzI1NiJ9.eyJhdWQiOiIyMDIzNDQwMzcxMzEiLCJthisisafaketokenm93bmVyIiwiZXhwIjoxNzY1ODE2NDQ0LCJpYXQiOjE3MzQyODA0NDQsImp0aSI6ImQ3asdfasdfasdfasdfasdfasdfasdfasdfWIyLTExMzA1MTBmZjFlOSIsInVzZXJuYW1lIjoiamdhc3NlbnNAZ21haWwuY29tIn0.rmbpyyeWKylwqEVyxBtpniPew0Mf_x9e4soCu1ti0JwoDZBLtBqGeoQJ_qVqxzhacUMBXdRgN7hITPa1BWo5PAthis-is-a-fake-token-fake-fake-token
> 15/12/2024 10:45:15 Connected to core-mosquitto:1883
> 15/12/2024 10:45:15 Subscribed to MQTT_TOPIC: /envoy/json
> 15/12/2024 10:45:15  Exception fetching stream data: HTTPSConnectionPool(host='envoy.local', port=443): Max retries exceeded with url: /ivp/meters/readings (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7fb05ed820>: Failed to establish a new connection: [Errno 101] Network unreachable'))
> 15/12/2024 10:45:15  Exception fetching stream data: HTTPSConnectionPool(host='envoy.local', port=443): Max retries exceeded with url: /ivp/meters/readings (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7fb05ed4c0>: Failed to establish a new connection: [Errno 101] Network unreachable'))

I assume that's an issue with my token. If so, how do i check /data/token.txt?
If it isn't… what am i doing wrong?

sddgit, 2024-12-15:
Have you just set this up, and it’s never worked? You could try using the IP address instead of envoy.local.

vk2him, 2024-12-15:
Coincidentally I had a similar issue a few weeks ago on my firmware V5 system where the Ethernet cable had come loose from the Envoy making intermittent connections. Once I reseated the cable the error went away. I'd therefore test your connection to be sure.

jgassens, 2024-12-15:
Nope, never able to get it to work. I am able to ping the IP address of my Envoy from the terminal in HA, so it works. I've also tried the numerical IP address and it I still get the same error…

vk2him, 2024-12-16:
Ok, so a few things here - it's not a mqtt broker issue as the logs show it connected and subscribed ok:

15/12/2024 10:45:15 Connected to core-mosquitto:1883
15/12/2024 10:45:15 Subscribed to MQTT_TOPIC: /envoy/json

It can also connect and talk to the Envoy as the logs show the serial number and firmware version.

What's not clear is what you've entered into the Add-on's Configuration page. Can you please take a look at this issue and paste your configuration screenshot here after redacting anything important?

https://github.com/vk2him/Enphase-Envoy-mqtt-json/issues/38

I have old Firmware v5 so can't test the token etc, however for fault finding in you browser go to this url and advise what it does?

https://envoy.local/ivp/meters/readings