vk2him/Enphase-Envoy-mqtt-json #42: [Doc] Add missing items to readme.md for v7 users.

batkite, 2024-12-16:
Hi,

Like some others, when I switched to firmware v7.x I lost most of the sensors, cards... that were working with the v5.
I'm not a HA champion so I mostly rely on other people contribution to search and fix my issues.

Regarding v7 firmares, I think you should add in the the readme this section :

```
mqtt:
  sensor:
    - name: "mqtt_net"
      state_topic: "/envoy/json"
      unit_of_measurement: "W"
      qos: 0
      state_class: measurement
      device_class: power
      value_template: '{{ value_json[1]["activePower"] | round(0) | int(0) }}'

    - name: "mqtt_production"
      state_topic: "/envoy/json"
      qos: 0
      unit_of_measurement: "W"
      value_template: '{{ value_json[0]["activePower"] | round(0) | int(0)}}'
      state_class: measurement
      device_class: power

    - name: "mqtt_consumption"
      state_topic: "/envoy/json"
      unit_of_measurement: "W"
      qos: 0
      state_class: measurement
      device_class: power
      value_template: '{{ ( value_json[1]["activePower"] + value_json[0]["activePower"] ) | round(0) | int(0) }}' # Net + production
```

With this, we recover the missing production sensor, and we get the Power Wheel card working with what's already in the readme examples.

[image]

My screenshot is showing duplicates but it's easier to mask something instead of trying to figure out why it's not there anymore.

vk2him, 2024-12-16:
Thanks for this - yes I'll add this and a few others to the readme

sddgit, 2024-12-16:
Here are a few more for your collection! These are formatted to be in a separate included mqtt.yaml file:

```
sensor:
  #####################
  # Envoy MQTT Sensors
  #####################

  - name: "mqtt_net"
    state_topic: "/envoy/json"
    unit_of_measurement: "W"
    qos: 0
    state_class: measurement
    device_class: power
    value_template: '{{ value_json[1]["activePower"] | round(0) | int(0) }}'
    
  - name: "mqtt_net_phase_a"
    state_topic: "/envoy/json"
    unit_of_measurement: "W"
    qos: 0
    state_class: measurement
    device_class: power
    value_template: '{{ value_json[1].channels[0].activePower | round(0) | int(0) }}'
    
  - name: "mqtt_net_phase_b"
    state_topic: "/envoy/json"
    unit_of_measurement: "W"
    qos: 0
    state_class: measurement
    device_class: power
    value_template: '{{ value_json[1].channels[1].activePower | round(0) | int(0) }}'
    
  - name: "mqtt_net_phase_c"
    state_topic: "/envoy/json"
    unit_of_measurement: "W"
    qos: 0
    state_class: measurement
    device_class: power
    value_template: '{{ value_json[1].channels[2].activePower | round(0) | int(0) }}'

  - name: "mqtt_production"
    state_topic: "/envoy/json"
    qos: 0
    unit_of_measurement: "W"
    value_template: '{{ value_json[0]["activePower"] | round(0) | int(0)}}'
    state_class: measurement
    device_class: power
    
  - name: "mqtt_production_ph_a"
    state_topic: "/envoy/json"
    qos: 0
    unit_of_measurement: "W"
    value_template: '{{ value_json[0].channels[0].activePower | round(0) | int(0)}}'
    state_class: measurement
    device_class: power
    
  - name: "mqtt_production_ph_b"
    state_topic: "/envoy/json"
    qos: 0
    unit_of_measurement: "W"
    value_template: '{{ value_json[0].channels[1].activePower | round(0) | int(0)}}'
    state_class: measurement
    device_class: power
        
  - name: "mqtt_production_ph_c"
    state_topic: "/envoy/json"
    qos: 0
    unit_of_measurement: "W"
    value_template: '{{ value_json[0].channels[2].activePower | round(0) | int(0)}}'
    state_class: measurement
[... 105 lines cut ...]
    
  - name: "mqtt_current_ph_a"
    state_topic: "/envoy/json"
    value_template: '{{ value_json[1].channels[0].current | round(2)}}'
    qos: 0
    unit_of_measurement: "A"
    state_class: measurement
    device_class: current
    
  - name: "mqtt_current_ph_b"
    state_topic: "/envoy/json"
    value_template: '{{ value_json[1].channels[1].current | round(2)}}'
    qos: 0
    unit_of_measurement: "A"
    state_class: measurement
    device_class: current
    
  - name: "mqtt_current_ph_c"
    state_topic: "/envoy/json"
    value_template: '{{ value_json[1].channels[2].current | round(2)}}'
    qos: 0
    unit_of_measurement: "A"
    state_class: measurement
    device_class: current
```

vk2him, 2024-12-17:
Thanks all for the example templates - I've update the Readme with a link to this thread - I encourage others to add additional templating examples to this thread to help others. Thanks again

closed 2024-12-17 by vk2him