vk2him/Enphase-Envoy-mqtt-json #43: Docker Compose Image Pull

BastyJuice, 2025-02-14:
can you allow pull for docker compose i cant pull the image becouse acess denied

vk2him, 2025-02-14:
Hi, I'm unsure how to do that, are you able to advise how to enable?

BastyJuice, 2025-02-15:
i dont know that also or maybee you can relase it also on https://hub.docker.com/

BastyJuice, 2025-02-15:
enphase-envoy-mqtt Pulling enphase-envoy-mqtt Error Error response from daemon: Head "https://ghcr.io/v2/vk2him/enphase-envoy-mqtt-json/manifests/latest": denied

BastyJuice, 2025-02-15:
The is a Fork  from @xorguy that can be pulled but it is ARM64 Port i need AMD64 https://github.com/xorguy/Enphase-Envoy-mqtt-json/network

vk2him, 2025-02-16:
Sorry, I don't know anything about how to get this to work via Docker.
Have you seen this pull request as I think he has been able to get it working in Docker? Maybe try his fork?

https://github.com/vk2him/Enphase-Envoy-mqtt-json/pull/39

Maybe ablso ask @xorguy or @MKSG-MugunthKumar for help?

BastyJuice, 2025-02-16:
thank you i hope @xorguy will respond an make and amd64 image as well

BastyJuice, 2025-02-16:
done this now by my self thanks anyway

closed 2025-02-16 by BastyJuice

vk2him, 2025-02-16:
Can you paste the code here for how you got it to work in case it can help others in future?

BastyJuice, 2025-02-18:
There is no Code simple createt my own Docker Image and pulled it to my Github.